raspberrypi/pico-sdk #1511: Time to make PICO_NO_FPGA_CHECK the default?

arg08, 2023-09-14:
When exploring the resource usage of a program, I was surprised to find running_on_fpga() appearing as an actual function rather than being eliminated at build time, and traced that to the default being:

`// PICO_CONFIG: PICO_NO_FPGA_CHECK, Remove the FPGA platform check for small cod
e size reduction, type=bool, default=0, advanced=true, group=pico_runtime
#ifndef PICO_NO_FPGA_CHECK
#define PICO_NO_FPGA_CHECK 0
#endif
`
in pico/platform.h

Surely at this stage of the development with millions of developers out there working with real devices and only a handful of in-house developers maybe still using FPGA once in a blue moon, the time has come to at least flip the default to '1'.

I appreciate it's only 48 bytes, but I think the justification is more about this unexpected code turning up in people's binaries and wasting time while they track down what it's doing there.

Ideally the whole FPGA support would be expunged from the public code base as it's just a source of confusion/timewasting and no practical value to nearly all the users, but if it is still seen as useful for in-house use at least the compromise of turning it off by default would be an improvement.

kilograham, 2023-12-14:
seems reasonable

kilograham, 2024-08-09:
it is now the default

closed 2024-08-09 by kilograham