raspberrypi/pico-sdk #1515: Can't connect over serial on Windows when getchar_timeout_us is used

mlukaszek, 2023-09-20:
Hi,

I'm observing a strange issue in the code I am working on. I am developing on a Mac, where I see no such issue.

However, when I connected a programmed Pico to a Windows 11 system:
- The COM port is visible in Device Manager as COM4 and has no issue
- However, attempting to connect with PuTTY is unsuccesful
- That said, the rest of the code is clearly running, it looks like it's just the stdio redirection that was unsuccesful

With the same Pico and the same machine, the hello_world/usb sample works without issues and I see the output from the device in PuTTY.

I eventually narrowed it down to the part of my main loop that attempts to use serial input over stdio. When I commented out the code using `getchar_timeout_us(0)`, the device can be connected to in Windows, and the output can be seen. Obviously the input does not work then.

Is this a known problem? Am I doing something wrong?
The only difference between a working and non-working code is the `getchar_timeout_us(0);` line in the main loop.
stdio is initialised with the `stdio_usb_init()` call in the beginning of my program.

peterharperuk, 2023-09-20:
> However, attempting to connect with PuTTY is unsuccesful

What do you mean by this? Do you mean you just don't get any characters enter on putty on the Pico? Or is putty returning an error? (which would be odd). A program demonstrating the issue might be useful.

mlukaszek, 2023-09-20:
I attempted to add a call to `getchar_timeout_us` to the helloworld sample and I was unable to reproduce.
That got me thinking :)

My original description of the issue turned out to be misleading - PuTTY can actually open the port (I was tricked by not seeing any output initially), and I can see the stdio output. My code handling input still had problems. No matter what I typed on the keyboard, my program did not handle it.

I added even more debug code to the function that handles input and it is clearly an issue with my handling of CRLF.
The characters actually get to the device as I type - however on Mac pressing enter only sends LF (10), while on Windows it only sends CR (13). I had the assumption that I'd get a full CRLF pair on Windows, I was mistaken.
Sorry for wasting your time. I'll rework the code and I am pretty confident it will work.

closed 2023-09-20 by mlukaszek