raspberrypi/pico-sdk #1516: ungetc() Does not push character back to STDIO stack

mmmlinux, 2023-09-20:
Calling ungetc(c, stdio); does not result in c being the next character when I call getchar_timeout_us(0). here is some example code. 

`long parseInt()
{
  printf("parseint\r\n");
  long value = 0;
  char c = getchar_timeout_us(0);
  printf("%i\r\n",c);
  if(c<0)
  {
    return 0;
  }
  bool isNeg = (c == '-');
  if(isNeg)
    c = getchar_timeout_us(0);
  while(isdigit(c))
  {
    value = (value *10) + (c - '0');
    c = getchar_timeout_us(0);
    printf("%i\r\n",c);
  }
  if(isNeg)
    value = -value;
  if(c != -1)
  {
    printf("pushed back: %i\r\n",c);
    ungetc(c,stdin);
  }
  printf("parse done\r\n");
  return value;
}`

ai6aj, 2023-10-24:
ungetc() works fine with plain old getc(stdin):

```
while(1) {
      if ((ch = getc(stdin)) > 0) {
        printf("\nGot %c, trying to stick it back in the buffer with ungetc\n",ch);
        int ungetc_rv = ungetc(ch+1, stdin);
        printf("ungetc(%d) returned %d\n",ch+1,ungetc_rv);
        if (ungetc_rv != EOF) {
          int ch2 = getc(stdin);
          if (ch2 == ch+1) {
            printf("ungetc worked\n");
          } else {
            printf("ungetc failed, expected %d but got %d\n",ch+1,ch2);
          }
        } else {
          printf("ungetc returned EOF\n");
        }
      }
    }
```

Produces:

```
a
Got a, trying to stick it back in the buffer with ungetc
ungetc(98) returned 98
ungetc worked
```

ungetc completely fails when getchar_timeout_us() is used.  If ungetc() is unable to stuff a byte back into the buffer it *should* report EOF, instead it cheerfully reports success even though it's done nothing:

```
aaa # NOTE: I blasted 3 bytes into the buffer so getchar_timeout_us() would have a character to report instead of EOF
Got a, trying to stick it back in the buffer with ungetc
ungetc(98) returned 98  # this should be -1, EOF
ungetc failed, expected 98 but got 97
```
I suspect that what's happening is that ungetc() *is* stuffing a character into stdin, but getchar_timeout_us() doesn't really read from stdin as the documentation claims.  Rather it reads from whatever underlying layer is *attached* to stdin by default, which is a subtle but important difference, especially if you've reopened stdin as something else.

More concerning to me is that if getc(stdin) is replaced in the above example with the [supposed *exact* equivalent getchar()](https://en.cppreference.com/w/c/io/getchar) , ungetc only works sometimes and output buffering seems to get very confused.

kilograham, 2024-08-09:
You can set PICO_STDIO_SHORT_CIRCUIT_CLIB_FUNCS=1 in your target_compile_options if you want to mix printf and fprintf(stdout). 

We will add ungetc to our short circuits in the future

closed 2024-08-09 by kilograham