raspberrypi/pico-sdk #1520: pioasm: Should the .origin directive change the label values accordingly.

mydana, 2023-10-21:
Am I missing something?

.origin is defined as
```Optional directive to specify the PIO instruction memory offset at which the program must load. Most commonly this is used for programs that must load at offset 0, because they use data based JMPs with the (absolute) jmp target being stored in only a few bits.```

So far, so good. But if the .origin is defined as something other than 0, wouldn't relative JMP targets be altered so that the relative JMP targets work in the new .origin?

The pioasm doesn't seem to work this way, and that puzzles me.

------------------------------------------------

Without an .origin directive jumps to address 0 as expected:

```
.program a_test
a_label:
	jmp a_label
```

 $ ./pioasm -o hex a_test.pio
0000

------------------------------------------------

But with an .origin directive it STILL jumps to address 0, and that doesn't make sense to me.

```
.program b_test
.origin 9
b_label:
	jmp b_label
```

$ ./pioasm -o hex b_test.pio
0000

I would think that the program loaded in offset 9, as in this example, that the JMP target would also be offset to 9, and wouldn't the example output be 0009.

Am I missing something, or is this a bug in the assembler?

mydana, 2023-10-21:
The JMP section in the data sheet says,

‘’’’Address: Instruction address to jump to. In the instruction encoding this is an absolute address within the PIO instruction memory.
‘’’’

If this is so, then this is a bug in the assembler. The JMP targets should be adjusted by an offset .origin.

lurch, 2023-10-21:
I've not written any PIO assembler myself, but I did a bit of rummaging in the documentation and 3.4.2.3. in https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf says:
"Note that because the PIO JMP instruction uses absolute addresses
in the PIO instruction memory, JMPs need to be adjusted based on the program load offset at
runtime. This is handled for you when loading a program with the SDK, but care should be taken when
encoding JMP instructions for use by OUT EXEC"

Does that explain what you're seeing?

mydana, 2023-10-21:
It does make sense. Much appreciated.

closed 2023-10-21 by mydana