raspberrypi/pico-sdk #1523: USB MSC has some sort of race condition

byteit101, 2023-10-23:
Since enabling a simple MSC drive, I have gotten frequency crashes. Using tinyusb 0.12, I got this log sequence, culminating in an assertion failure:
```
...snip...
USBD Setup Received 80 06 00 01 00 00 40 00 
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
  Short packet on buffer 0 with 0 bytes
Completed transfer of 0 bytes on ep 0 out
USBD Xfer Complete on EP 00 with 0 bytes
BUS RESET
USBD Bus Reset : Full Speed
  Short packet on buffer 0 with 18 bytes
Completed transfer of 18 bytes on ep 0 in

USBD Setup Received 00 05 1E 00 00 00 00 00 
  Set Address
USBD Xfer Complete on EP 80 with 18 bytes
usbd_control_xfer_cb 171: ASSERT FAILED
```
In contrast, successful Set Address looks like
```
...snip

USBD Setup Received 00 05 10 00 00 00 00 00 
  Set Address
  Short packet on buffer 0 with 0 bytes
Completed transfer of 0 bytes on ep 0 in
USBD Xfer Complete on EP 80 with 0 bytes

...snip
```

The reordering, combined with the fact that it's highly irregular, sometimes failing for 5 reboots in a row, then succeeding for 50%, then succeeding for 10 reboots in a row, makes me think this is some kind of race condition in the MSC TUD driver.

The stack trace was usbd_control.c:171, called from usbd.c:562

I updated TinyUSB to 0.15.0, and now get:
```
*** PANIC ***

ep 0 in was already available
```

byteit101, 2023-11-18:
I have uploaded a reproducible case as https://github.com/byteit101/test-tinyusb-pico
I have confirmed it with G++ 8.3 and G++ 12.2 (the latter requires uncommenting `target_link_options(${PROJECT_NAME} PRIVATE "LINKER:--no-warn-rwx-segments")` in the CMake file)
I have confirmed it on Debian 10, Linux pollux 5.10.0-21-amd64 #1 SMP Debian 5.10.162-1 (2023-01-21) x86_64 GNU/Linux
I haven't seen it on a "blank" flash.
I have seen it once the storage is formatted as `sudo mkfs.vfat -F12 -n "TinyUSB Tst" -S 4096 /dev/sdYOURDRIVE`
It appears to be linked to writing, but I can't seem to confirm this

liamfraser, 2024-08-01:
We'll be moving to a new TinyUSB release soon so once this is released can you check if this issue is still present?

kilograham, 2024-08-09:
we have updated TinyUSB with SDK 2.0.0 so please retry

closed 2024-08-09 by kilograham