raspberrypi/pico-sdk #1526: Assertion error when calling cyw43_arch_init() in FreeRTOS

ArcaEge, 2023-10-28:
When I try to call `cyw43_arch_init()` in a FreeRTOS task I get the following error printed to the serial:
```
assertion "get_core_num() == async_context_core_num(cyw43_async_context)" failed: file "......../pico-sdk/src/rp2_common/pico_cyw43_driver/cyw43_driver.c", line 53, function: cyw43_irq_init
```
I have the code on my repo [here](https://github.com/ArcaEge/globe-radio), however it is in a bit of a mess so here's a cleaned up version of the main file:
```
#include "FreeRTOS.h"
#include <task.h>
#include "pico/cyw43_arch.h"
#include <stdio.h>
#include "pico/stdlib.h"

void wifiConnectTask(void *pvParameters) {
    if (cyw43_arch_init()) {
        printf("failed to initialise\n");
        return;
    }
    cyw43_arch_enable_sta_mode();
    printf("Connecting to Wi-Fi...\n");
    if (cyw43_arch_wifi_connect_timeout_ms(WIFI_SSID, WIFI_PASSWORD, CYW43_AUTH_WPA2_AES_PSK, 30000)) {
        printf("failed to connect.\n");
        exit(1);
    } else {
        printf("Connected.\n");
    }
    while (true) {
        vTaskDelay(500);
    }
}

int main() {
    stdio_init_all();

    TaskHandle_t wifiHandle;

    xTaskCreate(wifiConnectTask, "wifiConnect", configMINIMAL_STACK_SIZE, NULL, 1, &wifiHandle);

    // Note: tried with and without this line, no difference
    vTaskCoreAffinitySet(wifiHandle, 1);
    
    vTaskStartScheduler();

    while (true)
        vTaskDelay(500);

    cyw43_arch_deinit();
    return 0;
}
```
Possibly related to #1478?

In addition, I've tried this with both the master and develop branches of the SDK.

ArcaEge, 2023-10-28:
The only workaround I've found so far is commenting out the `assert` statements.

kilograham, 2023-10-29:
Is this with the latest FreeRTOS where they have merged the `smp` branch into `main`... might be worth trying before that happened (it is possible something is broken as a result of the FreeRTOS merge)

peterharperuk, 2023-10-30:
Thanks for the code. I'll have a look. It would be useful to know the top commit of Freertos you're using.

peterharperuk, 2023-10-30:
I can reproduce the assert with the freertos commit you're using 37678b06568fda7aaff090b77e8e34d6ab29ede4, going back to the smp branch and it works ok. So as @kilograham says, it looks like something has changed.

peterharperuk, 2023-10-30:
The problem is that they renamed configNUM_CORES to configNUMBER_OF_CORES, you either need to add #define configNUM_CORES configNUMBER_OF_CORES to the freertos config file or do a search and replace in the pico-sdk, until we can fix this.

ArcaEge, 2023-10-30:
Thanks, it works after adding the #define to the config. I'm on the develop branch anyway due to #737 (has been fixed in #1488) and should get the fix when @kilograham approves the merge.

closed 2023-10-30 by ArcaEge