raspberrypi/pico-sdk #1527: waveshare RP2040 lcd 1.28 incorrect pin assignments

u6bkep, 2023-10-29:
[src/boards/include/boards/waveshare_rp2040_lcd_1.28.h](https://github.com/raspberrypi/pico-sdk/blob/master/src/boards/include/boards/waveshare_rp2040_lcd_1.28.h) pin assignments are incorrect. correct pin assignments can be found in [DEV_Config.h](https://github.com/waveshareteam/RP2040-Touch-LCD-1.28/blob/main/c/lib/Config/DEV_Config.h) from waveshare.

specifically I have found that `PICO_DEFAULT_SPI_CSN_PIN`, and `WAVESHARE_RP2040_LCD_RST_PIN` are incorrect, but there may be others.

kilograham, 2023-12-14:
if you have a PR to fix, then great

u6bkep, 2023-12-14:
that is my intention. that project has temporarily stalled due to work, but I might have some time for it in the upcoming holidays.

u6bkep, 2023-12-31:
I have discovered the problem is that the board in the pico-sdk is the `WAVESHARE_RP2040_LCD` and the board I have is the touch screen variant with a slightly different pin assignment. So I am [preparing a pull request](https://github.com/u6bkep/pico-sdk/tree/waveshare_128_touch) to add this board. The [schematic](https://files.waveshare.com/upload/3/3c/RP2040-Touch-LCD-1.28.pdf) for the touch variant lists a `W25Q128JVSIQ` as the flash chip, but the code defines a macro of `PICO_BOOT_STAGE2_CHOOSE_W25Q080`. Do I need to do anything special or can I just leave it as is?

lurch, 2024-07-24:
@kilograham / @Wren6991 I know nothing about the compatibility between different flash chips and different BOOT_STAGE2 files - could you respond to the question in the previous comment please?

kilograham, 2026-11-10:
fix merged into develop

closed 2026-11-10 by kilograham