raspberrypi/pico-sdk #1529: macos: ArmMbed/homebrew-formulae is deprecated

Gadgetoid, 2023-10-30:
The recommended install method for an ARM embedded toolchain is:

```
brew install --cask gcc-arm-embedded
```

This will probably affect the CI workflows:

https://github.com/raspberrypi/pico-sdk/blob/6a7db34ff63345a7badec79ebea3aaef1712f374/.github/workflows/macOS.yml#L25-L27

See: https://github.com/ARMmbed/homebrew-formulae#deprecation-notice

Using `brew install arm-none-eabi-gcc` results in builds breaking with:

```
arm-none-eabi-gcc: fatal error: cannot read spec file 'nosys.specs': No such file or directory
```

kilograham, 2023-12-14:
closing, as good for reference, but there are no instructions in the SDK

closed 2023-12-14 by kilograham

aallan, 2023-12-14:
Documentation around this should be fixed in the upcoming docs release.

nathan-contino, 2023-12-14:
Fixed!