raspberrypi/pico-sdk #1533: TraceAlyzer on pico running FreeRTOS

xannik, 2023-11-01:
Hello!

I am trying to integrate TraceAlyzer for my pico where I am running FreeRTOS. However, following the instructions from percepio [(getting started with freeRTOS)](https://percepio.com/getstarted/latest/html/freertos.html) I end up with some compilation errors which seems to be related to the #define __force_inline in platform.h.

The most important configuration for tracealyzer is done in the trcHeader.c from their TraceRecorder Repo, where they need the CMSIS headers

I have tried using RP2040.h or core_cm0plus.h but I see the same errors with both. (some more errors when using core_cm0plus.h).

I have also tried a third option where I dont provide any CMSIS headers and set the following:

```
#define TRC_CFG_HARDWARE_PORT TRC_HARDWARE_PORT_HWIndependent
```
```
src/rp2_common/pico_platform/include/pico/platform.h:#ifndef __always_inline
src/rp2_common/pico_platform/include/pico/platform.h:#define __always_inline __attribute__((__always_inline__))
src/rp2_common/pico_platform/include/pico/platform.h:#define __force_inline inline __always_inline
src/rp2_common/pico_platform/include/pico/platform.h:#define __force_inline  __always_inline
```
errors:
```
/tmp/pico-sdk/src/rp2_common/hardware_base/include/hardware/address_mapped.h:93:15: error: expected ';' before 'static'                                                                                     
   93 | __force_inline static uint32_t xip_alias_check_addr(const void *addr) {                                                                                                                             
      |               ^~~~~~~                                                                                                                                                                               
      |               ;                                                                                                                                                                                     
/tmp/pico-sdk/src/rp2_common/hardware_base/include/hardware/address_mapped.h:121:15: error: expected ';' before 'static'                                                                                    
  121 | __force_inline static void hw_set_bits(io_rw_32 *addr, uint32_t mask) {                                                                                                                             
      |               ^~~~~~~                                                                                                                                                                               
      |               ;                                                                                                                                                                                     
/tmp/pico-sdk/src/rp2_common/hardware_base/include/hardware/address_mapped.h:131:15: error: expected ';' before 'static'                                                                                    
  131 | __force_inline static void hw_clear_bits(io_rw_32 *addr, uint32_t mask) {                                                                                                                           
      |               ^~~~~~~                                                                                                                                                                               
      |               ;                                                                                                                                                                                     
/tmp/pico-sdk/src/rp2_common/hardware_base/include/hardware/address_mapped.h:141:15: error: expected ';' before 'static'                                                                                    
  141 | __force_inline static void hw_xor_bits(io_rw_32 *addr, uint32_t mask) {                                                                                                                             
      |               ^~~~~~~                                                                                                                                                                               
      |               ;                                                                                                                                                                                     
/tmp/pico-sdk/src/rp2_common/hardware_base/include/hardware/address_mapped.h:157:15: error: expected ';' before 'static'                                                                                    
  157 | __force_inline static void hw_write_masked(io_rw_32 *addr, uint32_t values, uint32_t write_mask) {                                                                                                  
      |               ^~~~~~~                                                                                                                                                                               
      |               ;                                                                                                                                                                                     
/tmp/pico-sdk/src/rp2_common/hardware_sync/include/hardware/sync.h:118:15: error: expected ';' before 'static'                                                                                              
  118 | __force_inline static void __sev(void) {                                                                                                                                                            
      |               ^~~~~~~                                                                                                                                                                               
      |               ;                                                                                                                                                                                     
/tmp/pico-sdk/src/rp2_common/hardware_sync/include/hardware/sync.h: In function '__sev':                                                                                                                    
/tmp/pico-sdk/src/rp2_common/hardware_sync/include/hardware/sync.h:119:5: warning: implicit declaration of function 'pico_default_asm_volatile' [-Wimplicit-function-declaration]                           
  119 |     pico_default_asm_volatile ("sev");                                                                                                                                                              
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                                                                       
/tmp/pico-sdk/src/rp2_common/hardware_sync/include/hardware/sync.h: At top level:                                                                                                                           
/tmp/pico-sdk/src/rp2_common/hardware_sync/include/hardware/sync.h:130:15: error: expected ';' before 'static'                                                                                              
  130 | __force_inline static void __wfe(void) {                                                                                                                                                            
      |               ^~~~~~~                                                                                                                                                                               
      |               ;
```
I am using a docker container for compilation with gcc version: gcc (Debian 12.2.0-14) 12.2.0 which I believe should support __attribute__((__always_inline__)).

```
FROM debian:bookworm-slim
RUN apt-get update && apt-get install -y --no-install-recommends \
    build-essential cmake gcc-arm-none-eabi libnewlib-arm-none-eabi \
    libstdc++-arm-none-eabi-newlib git ca-certificates make python3 \
    libusb-1.0-0-dev pkg-config clangd clang-format picocom libncurses5-dev \
    libncursesw5-dev wget tar libncurses5
```
CMake structure is one main file in repo root: 

```
cmake_minimum_required(VERSION 3.13)

[... 22 lines cut ...]
pico_sdk_init()

add_executable(${PROJECT_NAME}
        main.c
)

[... cut ...]
target_link_libraries(${PROJECT_NAME}
        pico_stdlib
        pico_stdio_usb
        hardware_pwm
        hardware_clocks
        FreeRTOS-Kernel-Heap4
        cmsis_core)

pico_add_extra_outputs(${PROJECT_NAME})
```
If I copy the platform.h file into my project and directly use `__attribute__((__always_inline__))` instead of `__force_inline` then I don't see those errors

am I missing something or what could cause this? 

Have you been able to setup tracealyzer on a pico running freeRTOS?

Much appreciation for the help!

peterharperuk, 2023-11-02:
I see someone has answered this question on the forum. Does that info resolve the problem? https://forums.raspberrypi.com/viewtopic.php?t=358714

xannik, 2023-11-03:
Hello! Yes it resolved this issue but created some arm-none-eabi ld problems instead. Currently trying to figure out why that happend.

xannik, 2023-11-03:
It is working now. Solution can be found here: https://forums.raspberrypi.com/viewtopic.php?t=358714

closed 2023-11-03 by xannik