raspberrypi/pico-sdk #1537: Cmake build

paulhamsh, 2023-11-08:
Hi    
I want to use the HCI layer for the Pico W without BTStack - just using the cyw43-driver.   
I need to use the following functions:   

cyw43_init()    
cyw43_bluetooth_hci_write()    
cyw43_bluetooth_hci_read()    

Can anyone help with what to put into a cmake file to build this correctly. The right library and build options?    

I have used these functions in a Micropython C module (https://github.com/paulhamsh/HCI-PicoW) but don’t understand cmake and the Pico build process enough to get this to work in a standalone program.    

Thanks

paulhamsh, 2023-11-08:
So, this works but isn't optimal because it includes btstack and I had to create a btstack_config.h file.

```
cmake_minimum_required(VERSION 3.13)

include(./pico_sdk_import.cmake)

project(my_project C CXX ASM)
set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)
set(PICO_BOARD pico_w)
set(CYW43_ENABLE_BLUETOOTH 1)

pico_sdk_init()

add_executable(hci_pico main.c )
target_link_libraries(hci_pico pico_stdlib   pico_cyw43_arch_none   pico_btstack_ble   pico_btstack_cyw43)
target_include_directories(hci_pico PRIVATE ${CMAKE_CURRENT_LIST_DIR}) # bt stack config
pico_enable_stdio_usb(hci_pico 1)
pico_enable_stdio_uart(hci_pico 0)
pico_add_extra_outputs(hci_pico)
```

peterharperuk, 2023-11-09:
Linking to pico_btstack_ble will pull in the btstack library. cyw43_bluetooth_hci_write and cyw43_bluetooth_hci_read are in cyw43_driver. You would have to set CYW43_ENABLE_BLUETOOTH=1 yourself?

paulhamsh, 2023-11-09:
Thank you - no error. Where is the forum?

peterharperuk, 2023-11-09:
https://forums.raspberrypi.com/viewforum.php?f=145

paulhamsh, 2023-11-11:
Resolved thank you

closed 2023-11-11 by paulhamsh