raspberrypi/pico-sdk #1542: pico/util/queue.h No protection for NULL pointer before copying data.

Bibi2424, 2023-11-10:
I don't think this is a big problem for `queue_add_internal` and `queue_peek_internal`, the user can be expected to provide valid pointer for those.
But I think there should be a protection for `queue_remove_internal`.

The best example is that if I want to first call `queue_peek_internal`, I already have a copy of the data and if I then simply want to remove the entry from the queue, I don't want to have to pass a valid address. I simply want to call `queue_remove_internal(q, NULL, ...)` to remove the entry from the queue.

**Example:**
```
queue_t q;
queue_init(&q, sizeof(uint), 8);
...
uint data;
queue_try_peek(&q, &data);
if(data == 0) {
    // This wil cause a problem. However, I already have the data at that point.
    // I would expect being able to remove it from the queue without a valid pointer
    queue_try_remove(&q, NULL); 
}
```

peterharperuk, 2023-11-10:
You mean if you don't care about the entry that was removed you should be able to pass NULL?

Bibi2424, 2023-11-10:
Yes.
The other thing would be to simply prevent an error if the user wants to pass a NULL for any other reason. This argument is also valid for the other functions `queue_add_internal` and `queue_peek_internal`.

kilograham, 2024-05-19:
seems like a reasonable request

peterharperuk, 2024-08-09:
You can pass NULL to queue_remove_internal and queue_peek_internal now

closed 2024-08-09 by peterharperuk