raspberrypi/pico-sdk #1543: Getting started doc corrections for MAC

gljubojevic, 2023-11-11:
In Getting Started https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf
Build tools on MacOS Sonoma, didn't work, couldn't build anything, page 37: 
```bash
$ brew install cmake
$ brew tap ArmMbed/homebrew-formulae
$ brew install arm-none-eabi-gcc
```
corrected like this:
```bash
$ brew uninstall arm-none-eabi-gcc
$ brew auto remove
$ brew install --cask gcc-arm-embedded
```
This probably requires another test and review...

When building openOCD, install step is missing, page 63, commands:
```bash
$ cd ~/pico
$ git clone https://github.com/raspberrypi/openocd.git --branch rp2040 --depth=1
$ cd openocd
$ export PATH="/usr/local/opt/texinfo/bin:$PATH" ①
$ ./bootstrap
$ ./configure --disable-werror ②
$ make -j4
```
You should add:
```bash
$ make install
```
Missing install step causes vscode not finding openocd and/or required scripts for debug.

Debugging in VSCode using pico probe (page:65) needed to add in launch.json copy from (launch-probe-swd.json):
```javascript
"openOCDLaunchCommands": ["adapter speed 5000"],
```
Setting is taken from better way to test working openocd when pico probe is connected
```bash
$ openocd -f interface/cmsis-dap.cfg -c "adapter speed 5000" -f target/rp2040.cfg -s tcl
```
also needed to change to go along with arm tools:
```javascript
 "gdbPath" : "gdb-multiarch",
```
to:
```javascript
"gdbPath" : "arm-none-eabi-gdb",
```

So final launch.json for MAC is:
```json
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Pico Debug",
            "cwd": "${workspaceRoot}",
            "executable": "${command:cmake.launchTargetPath}",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
	    "openOCDLaunchCommands": ["adapter speed 5000"],
            // This may need to be "arm-none-eabi-gdb" for some previous builds
            "gdbPath" : "arm-none-eabi-gdb",
            "device": "RP2040",
            "configFiles": [
                // This may need to be "interface/picoprobe.cfg" for some previous builds
                "interface/cmsis-dap.cfg",
                "target/rp2040.cfg"
            ],
            "svdFile": "${env:PICO_SDK_PATH}/src/rp2040/hardware_regs/rp2040.svd",
            "runToEntryPoint": "main",
            // Work around for stopping at main on restart
            "postRestartCommands": [
                "break main",
                "continue"
            ]
        }
    ]
}
```

Hope this helps to improve Getting Started doc and other Mac users, now I can build and debug pico programs.

aallan, 2023-11-13:
Duplicate of raspberrypi/pico-feedback#355

closed 2023-11-13 by aallan

Dygear, 2023-11-21:
Just an FYI `make install` after building openocd from the above commands didn't quite work out for me. Needs to be a `sudo make install`.