raspberrypi/pico-sdk #1545: CMake Deprecation Warning: cmake_minimum_required

gmt3141, 2023-11-13:
Recently, when building `pioasm` and/or `elf2uf2`, CMake warns about `cmake_minimum_required`:

```
[ 96%] Performing configure step for 'PioasmBuild'
loading initial cache file /home/gmt/Projects/pico/build/pico-sdk/src/rp2_common/pico_cyw43_driver/pioasm/tmp/PioasmBuild-cache-Release.cmake
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
```
[https://github.com/raspberrypi/pico-sdk/pull/1477](This issue) reported the problem, but submitted path is not correct.

gmt3141, 2023-11-14:
https://github.com/raspberrypi/pico-sdk/pull/1546

peterharperuk, 2023-11-20:
I can't seem to reproduce this. Do you still have the problem if you delete your build folder and start from scratch?

kilograham, 2023-12-15:
merged into develop

PR #1546 (2023-12-15): CMake cmake_minimum_required Deprecation Warning
Fixes: #1545 

Recently, when building pioasm and/or elf2uf2, CMake warns about cmake_minimum_required:

```
[ 96%] Performing configure step for 'PioasmBuild'
loading initial cache file /home/gmt/Projects/pico/build/pico-sdk/src/rp2_common/pico_cyw43_driver/pioasm/tmp/PioasmBuild-cache-Release.cmake
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the p [... cut]
closed 2023-12-15 by kilograham