raspberrypi/pico-sdk #1551: Move SPINLOCK registers to separate section is SVD register map

stefan-misik, 2023-11-17:
Hello
I (https://github.com/raspberrypi/picoprobe/issues/109) and other people (https://forums.raspberrypi.com/viewtopic.php?t=331421#p2027769) have experienced hard-to-debug problem (for a RP2040 novice at least) where the program would refuse to advance past any spin-lock while debugging.

This is caused by the debugger inadvertently acquiring all the spin-locks any time you try to e.g. manipulate the GPIO registers from the graphical debugging session. This is caused by debugger reading all registers in the SIO group (which includes spin-locks) when the group is expanded.

Would it maybe make sense to move the spin-locks to different section of SVD file to prevent this issue?

Thanks.
Best Regards,
Stefan

lurch, 2023-11-17:
See also https://github.com/raspberrypi/pico-examples/issues/363

kilograham, 2024-05-19:
This is a long-standing issue, and should be fixed in the next release (by annotating the SVD correctly that reads have side effects)

kilograham, 2024-08-09:
these now include read sideeffects in the SVD so should be fine if your debugger supports

closed 2024-08-09 by kilograham