raspberrypi/pico-sdk #1555: `alarm_pool_add_alarm_at` fires when missed even if interrupts are disabled

yacinbm, 2023-11-21:
**Context**
I have an application that queues alarms at different timestamps, which can be very small depending on the context. However, before calling `alarm_pool_add_alarm_at` I disable interrupts with the `save_and_disable_interrupts` API. 

**Expected behaviour.**
When disabling interrupts, I expect that my alarm callback cannot be called until the interrupts are re-enabled. However, in [`alarm_pool_add_alarm_at()`](https://github.com/raspberrypi/pico-sdk/blob/6a7db34ff63345a7badec79ebea3aaef1712f374/src/common/pico_time/time.c#L247-L250), the callback is called if the timestamp was missed, regardless of state of the interrupts. 

**Possible Fix**
In [`add_alarm_under_lock`](https://github.com/raspberrypi/pico-sdk/blob/6a7db34ff63345a7badec79ebea3aaef1712f374/src/common/pico_time/time.c#L114-L118), when `is_missed` is true, check the state of the interrupts. If the interrupt level of the alarm is currently blocked, don't remove the entry from the pool and set the timer IRQ. This way, when interrupts are re-enabled, the timer IRQ will fire and the callback will be called without bypassing the interrupt block.

kilograham, 2024-05-19:
Planning a new `pico_time` implementation which should not suffer this problem

kilograham, 2024-08-09:
this is fixed

closed 2024-08-09 by kilograham