raspberrypi/pico-sdk #1557: Function Naming for static void uart_set_irq_enables() - Inconsistent in SDK

drankinatty, 2023-11-27:
The naming for `static void uart_set_irq_enables()` is horribly inconsistent with the remaining functions in the SDK. All other similar functions are named `..._enabled()`, but this one function has an `'s'` where the `'d'` normally is. 

For example see the [Pico UART SDK Documentation Page](https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#rpipf53ff9493512b6885dbc) and compare:

* [uart_set_irq_enables](https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#ga8c0c74c65b12281345886829a15c031f)
* [uart_is_enabled](https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#ga6397bfad56e634a2ab31b6953a44caa7)
* [uart_set_fifo_enabled](https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#gac3daee9514dacfb76f3c04b131e2e41a)

Or just about any other Hardware API function naming in the SDK, e.g.

* [adc_irq_set_enabled](https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#gaf8e363e63f40e00c169f7355c9668681)

This is a naming issue, so it's not the end of the world, but it is a significant *Mental-Train Wreck* where you are reading though code a come across this one `"...enables()"`. Heaven knows the old `'s'`, `'d'` typo on a QWERTY keyboard isn't exactly unheard of...

The issue becomes how best to handle it? There is an argument to make that "it's been wonky this long, just leave it", but it is probably something that should be cleaned up. I'm not sure how best to do it other than rename it, set a `#define` to the old name and put out a deprecation notice that the old name may go away in a future release. You all know best.

Don't get me wrong, this isn't a complaint, the SDK is wonderful, just trying to do my part and pass along an opportunity to improve it. Keep up the great work.

peterharperuk, 2023-11-27:
It's unlikely to be a typo. It enables both rx and tx fifo interrupts as requested.

kilograham, 2024-05-19:
yeah; some things squeaked thru the naming consistency just before SDK1.0 ... 

This should really be 

`uart_set_irqs_enabled` and there should be `adc_set_irq_enabled`

generally we will just add a #define for the old spelling to the new correct one

peterharperuk, 2024-08-09:
uart_set_irqs_enabled now exists

closed 2024-08-09 by peterharperuk