raspberrypi/pico-sdk #1575: adc_read() not checking for error conditions

ruehlchris, 2023-12-06:
Hi,

while testing the ADC single read I become aware that the current implementation not check for error conditions, only
look for the ADC_CS_READY_BITS.

I suggest to change the return value from a uint32_t to int32_t and return a negative value if
ADC_CS_ERR_BITS or ADC_CS_ERR_STICKY_BITS had been raised.

Here is what I used in my implementation.

[int32_t_adc_read.patch.gz](https://github.com/raspberrypi/pico-sdk/files/13575534/int32_t_adc_read.patch.gz)

You may consider to change the implementation.

Chris

lurch, 2023-12-08:
@ruehlchris If you're suggesting a code-change, it's generally better to do that by submitting a Pull Request to the `develop` branch, than to attach a `.patch.gz` file to an Issue.

ruehlchris, 2023-12-11:
Thanks,

I will have pull request ready soon.

closed 2023-12-11 by ruehlchris