raspberrypi/pico-sdk #1578: CMAKE_PREFIX_PATH content lost in middle of the CMake configuration step

opilat, 2023-12-12:
Pico examples project (picow_blink) configured with _-DCMAKE_PREFIX_PATH=path_to_arm_none_eabi_toolchain_ command line option fails to configure. Before the C compiler test step, it loses the content of the CMAKE_PREFIX_PATH.  
`-- Detecting C compiler ABI info
CMake Error at C:/Projects/pico/pico-sdk/cmake/preload/toolchains/find_compiler.cmake:28 (message):
  Compiler 'arm-none-eabi-gcc' not found, you can specify search path with
  "PICO_TOOLCHAIN_PATH".`
[configure_issue.txt](https://github.com/raspberrypi/pico-sdk/files/13648643/configure_issue.txt)

Krakonos, 2023-12-12:
Note this issue can be fixed by either passing the `CMAKE_PREFIX_PATH` as an environment variable, or setting before running project:

```
set(CMAKE_TRY_COMPILE_PLATFORM_VARIABLES CMAKE_PREFIX_PATH)
```

The problem is that try try_compile creates a CMakeFiles.txt that includes the preload/toolchains/pico_arm_gcc.cmake. I'm not sure how this mechanism should work, but seems like bad practice to depend on a specific ways to pass the variable. If something, I suggest using the variable above to pass the previously found compiler paths directly, instead of trying to find it again - since there may be other factors causing a different compiler is found in the toplevel cmake, and a different one is found in the generated try-compile cmake.

will-v-pi, 2024-11-12:
Looks like this was fixed in the 2.0.0 release, by adding this line https://github.com/raspberrypi/pico-sdk/blob/efe2103f9b28458a1615ff096054479743ade236/cmake/pico_pre_load_toolchain.cmake#L7

Can you try the 2.0.0 release and confirm if this has fixed your issue?

kilograham, 2024-11-19:
closing as this should have been fixed; please re-open if still an issue

closed 2024-11-19 by kilograham