raspberrypi/pico-sdk #1579: pico/version.h missing on Raspberry Pi 4 after running ./pico_setup.h multiple times

brianlmerritt, 2023-12-13:
I have installed the ~/pico_setup.sh` script including sdk, examples, openocd etc on a brand new Raspberry Pi 4 image 

```
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
```

```
Revision	: d03115
Serial		: 100000005d820ddf
Model		: Raspberry Pi 4 Model B Rev 1.5
```

The c/c++ include configuration is

```
${workspaceFolder}/**
${env:PICO_SDK_PATH}/**
```

IntelliSense on VScode suggests everything is OK except the `#include "pico/stdlib.h"`

Hovering over this code, I get a cannot open source file "pico/version.h" (dependency of "pico/stdlib.h")C/C++(1696)

`locate pico/version.h` generates

```
/home/xxx/pico/pico-examples/build/generated/pico_base/pico/version.h
/home/xxx/pico/pico-sdk/src/common/pico_base/include/pico/version.h
/home/xxx/pico/pico-sdk/src/common/pico_base/include/pico/version.h.in
/home/xxx/pico/picoprobe/build/generated/pico_base/pico/version.h
```

I can get rid of the error by adding an extra include

`${HOME}/pico/pico-examples/build/generated/pico_base/pico/**`

But it seems like the `~/pico_setup.sh` is missing a build step

I have run this script a number of times but am missing some step or config I think

brianlmerritt, 2023-12-13:
Further thoughts

1. ``cmake ..`` from build directory and `make` run fine, so they are picking up the missing `version.h` from somewhere, but intellisense and debug are crapping out even though the same arm-none-eabi-gcc c++ compiler is used 
2. The VSCode settings were generated by the latest pico_project_generator
3. Adding the path to the already made version.h file as per above and I am now getting a no such file or directory error in `#include pico/stdlib.h` even though the file is located at `/home/blm/pico/pico-sdk/src/common/pico_stdlib/include/pico/stdlib.h`

Some VSCode config details:

.c_cpp_properties.json
```
{
    "configurations": [
        {
            "name": "Linux",
            "includePath": [
                "${workspaceFolder}/**",
                "${env:PICO_SDK_PATH}/**",
                "${HOME}/pico/pico-examples/build/generated/pico_base/pico/**"
            ],
            "defines": [],
            "compilerPath": "/usr/bin/arm-none-eabi-gcc",
            "cStandard": "gnu17",
            "cppStandard": "gnu++14",
            "intelliSenseMode": "linux-gcc-arm",
            "configurationProvider": "ms-vscode.cmake-tools"
        }
    ],
    "version": 4
}
```

extensions.json
```
{
  "recommendations": [
    "marus25.cortex-debug",
    "ms-vscode.cmake-tools",
    "ms-vscode.cpptools"
  ]
}

launch.json
```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Cortex Debug",
            "cwd": "${workspaceRoot}",
            "executable": "${command:cmake.launchTargetPath}",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "gdbPath": "gdb-multiarch",
            "serverArgs": [
                "-c",
                "adapter speed 5000"
            ],
            "device": "RP2040",
            "configFiles": [
                "interface/cmsis-dap.cfg",
                "target/rp2040.cfg"
            ],
            "svdFile": "${env:PICO_SDK_PATH}/src/rp2040/hardware_regs/rp2040.svd",
            "runToEntryPoint": "main",
            "postRestartCommands": [
                "break main",
                "continue"
            ]
        },
        {
            "name": "C/C++: arm-none-eabi-gcc build and debug active file",
            "type": "cppdbg",
            "request": "launch",
            "program": "${fileDirname}/${fileBasenameNoExtension}",
            "args": [],
            "stopAtEntry": false,
            "cwd": "${fileDirname}",
            "environment": [],
            "externalConsole": false,
            "MIMode": "gdb",
            "setupCommands": [
                {
                    "description": "Enable pretty-printing for gdb",
                    "text": "-enable-pretty-printing",
                    "ignoreFailures": true
                },
                {
                    "description": "Set Disassembly Flavor to Intel",
                    "text": "-gdb-set disassembly-flavor intel",
                    "ignoreFailures": true
                }
            ],
            "preLaunchTask": "C/C++: arm-none-eabi-gcc build active file",
            "miDebuggerPath": "/usr/bin/gdb"
        }
    ]
}
```

settings.json

```
{
  "cmake.configureOnOpen": false,
  "cmake.statusbar.advanced": {
    "debug" : {
      "visibility": "hidden"
              },
    "launch" : {
      "visibility": "hidden"
               },
    "build" : {
      "visibility": "hidden"
               },
    "buildTarget" : {
      "visibility": "hidden"
               },
     },
}
```

tasks.json
```
{
    "tasks": [
        {
            "type": "cppbuild",
            "label": "C/C++: arm-none-eabi-gcc build active file",
            "command": "/usr/bin/arm-none-eabi-gcc",
            "args": [
                "-fdiagnostics-color=always",
                "-g",
                "${file}",
                "-o",
                "${fileDirname}/${fileBasenameNoExtension}"
            ],
            "options": {
                "cwd": "${fileDirname}"
            },
            "problemMatcher": [
                "$gcc"
            ],
            "group": {
                "kind": "build",
                "isDefault": true
            },
            "detail": "Task generated by Debugger."
        }
    ],
    "version": "2.0.0"
}
```

lurch, 2023-12-13:
> cmake .. from build directory and make run fine

It sounds like this is **just** an intellisense configuration problem then? Perhaps https://github.com/raspberrypi/pico-feedback/issues/148 helps? I've also got a feeling that this has been mentioned in other issues too, but can't find them right now.
Perhaps somebody on the https://forums.raspberrypi.com/ will be able to give you some help with your VSCode configuration?

brianlmerritt, 2023-12-15:
I'm working on the intellisense side, but wanted to point out it only compiles if I add other include paths than the sdk.

What code or process should convert version.h.in to version.h

brianlmerritt, 2023-12-15:
If you want to replicate this, create a new Raspberry Pi 4 or 5 image, and run Chapter 1 of the install here https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf with ./pico_setup.sh

lurch, 2023-12-15:
ping @nathan-contino as I believe he looked at those instructions most recently?

nathan-contino, 2023-12-15:
We have a new version of the Getting Started guide coming out soon that should address these (and quite a few other) issues. Check back early next week and the new version should be available!

brianlmerritt, 2023-12-16:
Cool and thanks - I have a work around so you can either close the ticket and assume all will be well or leave it until the documentation catches up :D

lurch, 2023-12-30:
See also https://github.com/raspberrypi/pico-feedback/issues/211#issuecomment-1871879760