raspberrypi/pico-sdk #1586: Setting AP/STA mode resets PicoW during debug

HouseSimOn, 2023-12-19:
# Description

Last time I decided to try use PicoProbe to have a debugger. I'm using Pico as PicoProbe connected to a PicoW. After some troubles (for example with `sleep_ms` function in debug https://github.com/raspberrypi/picoprobe/issues/45) it works. Then I decided to load simple example with cyw43 driver to setup AP or to set STA mode. Unfortunately, call of the `cyw43_arch_enable_ap_mode` or `cyw43_arch_enable_sta_mode` functions cause reset of the Pico and debug session is interrupted. Program loaded to the PicoW without debugger works. 

First I've used `pico_cyw43_arch_lwip_threadsafe_background` but I thought that maybe multi-core under debug is a problem but no... `pico_cyw43_arch_lwip_poll` also fails.

## Reproduce

You can find my test example here: https://github.com/HouseSimOn/PicoSdkStartTemplate/blob/test/cyw43debug/Source/main.c - It's ready to compile and test.

## Output 

### Debug session

1. `cyw43_arch_enable_ap_mode("MyAPTest", "12345678", CYW43_AUTH_WPA2_AES_PSK);` is called:
[image]

2. `cyw43_arch_enable_ap_mode` is commented out:
[image]

### No debug

`cyw43_arch_enable_ap_mode("MyAPTest", "12345678", CYW43_AUTH_WPA2_AES_PSK);` is called - it works as expected. AP is created.

peterharperuk, 2023-12-20:
I debug pico-w all the time with a debug probe. I can't imagine how/why that call would cause a problem for the debugger.

peterharperuk, 2023-12-20:
Cannot reproduce. I ran your code via the debugger and nothing "reset"

peterharperuk, 2023-12-21:
I used linux I don't have a mac. Yes I used a Pico to do a debug. I suggest you ask on the forum, maybe they'll have better luck guessing what you're doing wrong.

HouseSimOn, 2023-12-22:
@peterharperuk I've performed test on Ubuntu 22.04.3 LTS. I've used `gdb-multiarch` so I had to change one line in `launch.json` to:
```json
"gdbPath" : "gdb-multiarch"
```
I've also added `set USE_CORE 0` to the `target/rp2040.cfg` file. And there is the same problem as on macOS. Reset after `cyw43_arch_enable_ap_mode` call. Everything work ok when this line is commented out.

Questions:
1. Have you added `set USE_CORE 0` to the `target/rp2040.cfg` file?
2. Do you use official `openocd` or from this repo: https://github.com/raspberrypi/openocd.git?
3. Is there anything else that should be configured/installed than:
   - install `openocd`, 
   - install `gdb-multiarch` or use `arm-none-eabi-gdb`, 
   - change rp2040.cfg?

I've created post on forum: https://forums.raspberrypi.com/viewtopic.php?t=361958 as you suggested.

peterharperuk, 2023-12-22:
Why are you setting USE_CORE? Does it work without that?

HouseSimOn, 2023-12-22:
See this issue: https://github.com/raspberrypi/picoprobe/issues/45
I had to setup USE_CORE because `sleep_ms` function breaks debug - or maybe I mixed something?

lurch, 2023-12-23:
Based on the comments in https://github.com/raspberrypi/picoprobe/issues/45#issuecomment-1332610961 (and having a quick search in the RP2040 datasheet) I guess setting `TIMER: DBGPAUSE.DBG1` to `0` might be an alternative to telling OpenOCD to only use core 0? 
(see https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2040/hardware_regs/include/hardware/regs/timer.h#L157 and https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2040/hardware_structs/include/hardware/structs/timer.h#L63 )

HouseSimOn, 2023-12-25:
@lurch Adding `timer_hw->dbgpause = 0;` to the code doesn't help, but it's true that it helps with `sleep_ms` issue. In this case we don't have to modify `target/rp2040.cfg` file.

I connected PicoProbe with PicoW as on the picture below (source: https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf):
[image]

I solved issue when I connected to the PicoW board USB cable (and disconnected VSYS between boards). It looks like there is no enough power on VSYS pin to power PicoW when AP/STA mode starts (any current-peak?). I have to investigate it. But maybe this information should be added to the **Picoprobe wiring** subsection in [Getting started with Raspberry Pi Pico](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf) document?

peterharperuk, 2024-02-14:
I couldn't seem to reproduce this. I had the usb connected to the pico debugger which is powering the pico_w as shown above. I could connect to wifi just fine when debugging.

closed 2024-02-14 by peterharperuk