raspberrypi/pico-sdk #1588: Support for HTTP GET and POST in Raspberry Pi Pico W SDK

Chrisx43, 2023-12-20:
I would like to ask about the current and future support for HTTP GET and POST methods in the Raspberry Pi Pico W SDK

Is there currently the capability or a plan to integrate HTTP GET and POST functionalities into the Raspberry Pi Pico W SDK? These features would be immensely beneficial for projects involving web communication, data retrieval, and integration with online services.

If support for HTTP communication is currently not available, I kindly ask  of adding this feature in future releases of the SDK. It would significantly expand the potential use cases and applications for the Raspberry Pi Pico W.

Additional Information:
- Are there any workarounds or third-party libraries that currently provide HTTP support for the Raspberry Pi Pico W?
- If HTTP support is planned for a future release, an estimated timeline or roadmap would be appreciated.

Thank you.

aallan, 2023-12-20:
Have you looked at the [raspberrypi/pico-examples](https://github.com/raspberrypi/pico-examples) repo? There are networking examples there, e.g. see https://github.com/raspberrypi/pico-examples/tree/master/pico_w/wifi/tls_client for how to make a HTTPS GET request.

peterharperuk, 2023-12-20:
There's a http API here - which I hope to get included in the next release

https://github.com/raspberrypi/pico-sdk/pull/1388
https://github.com/raspberrypi/pico-examples/pull/385

Feedback welcome.

closed 2023-12-20 by peterharperuk

Chrisx43, 2023-12-20:
Hey, thanks for the help! I  clone the branch with the http_client example. However, I'm hitting a wall trying to build the http_client, it builds sucessfully but no files in the build folder. Strangely, I can build the other examples in the wifi folder just fine.

Quick question: Is there also a POST function available? I can't check it myself at the moment because there's no definition for the header files in http_client.c, which I suspect is causing the build failure.

Any tips on fixing the build issue and info on the POST function would be awesome

peterharperuk, 2023-12-21:
Sorry, I just checked and it builds fine for me. Suggest you start from a clean folder. It doesn't support POST, we're just using the http client that comes from lwip.

chseAtHms, 2023-12-21:
"Is there any possibility that the Pico will soon support POST too?"

lurch, 2023-12-21:
I did a quick bit of searching and just found https://github.com/lwip-tcpip/lwip/pull/14
(although I've not tested it mysefl)