raspberrypi/pico-sdk #1592: FreeRTOS with LWIP

hajmedwissem, 2023-12-23:
hi ,
I use LWIP, which is part of my Pico SDK. I tried to integrate LWIP with FreeRTOS, but I encountered errors in `lwip_port_rand`.
[build] /usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/pico_server.dir/pico-sdk/lib/lwip/src/core/dns.c.obj: in function `dns_create_txid':
[build] /home/wissem/Desktop/pico_server/pico-sdk/lib/lwip/src/core/dns.c:1025: undefined reference to `lwip_port_rand'
[build] /usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/pico_server.dir/pico-sdk/lib/lwip/src/core/tcp.c.obj: in function `tcp_init':
[build] /home/wissem/Desktop/pico_server/pico-sdk/lib/lwip/src/core/tcp.c:204: undefined reference to `lwip_port_rand'
[build] /usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/pico_server.dir/pico-sdk/lib/lwip/src/core/udp.c.obj: in function `udp_init':
[build] /home/wissem/Desktop/pico_server/pico-sdk/lib/lwip/src/core/udp.c:90: undefined reference to `lwip_port_rand'

"I think the issue is in the configuration files of FreeRTOS and lwIP. Can someone assist me in configuring both FreeRTOS and lwIP to work together?"

peterharperuk, 2023-12-24:
We already have freertos working with lwip. See pico-examples/pico_w/wifi/freertos

hajmedwissem, 2023-12-24:
Thank you, Peter.

I apologize; maybe I was not clear in the details of my question. I want to send data over USB and integrate FreeRTOS, LWIP, and TinyUSB to work together. So, I don't need to use the CYW43 chip.

peterharperuk, 2024-01-12:
I think this must be a bug in your code. Make sure you link to pico_lwip_freertos. This should add pico_lwip/include to the include path. arch/cc.h is included via something like this

pico-sdk/src/rp2_common/pico_lwip/include/arch/cc.h
pico-sdk/lib/lwip/src/include/lwip/arch.h
pico-sdk/lib/lwip/src/include/lwip/debug.h
pico-sdk/lib/lwip/src/include/lwip/opt.h
pico-sdk/lib/lwip/src/core/dns.c

We define LWIP_RAND in here

#ifndef LWIP_RAND
#include "pico/rand.h"
// Use the pico_rand library which goes to reasonable lengths to try to provide good entropy
#define LWIP_RAND() get_rand_32()
#endif
#endif /* __CC_H__ */

So you should ask why this isn't working for your code.

peterharperuk, 2024-01-12:
Alternatively, just define lwip_port_rand yourself to be get_rand_32