raspberrypi/pico-sdk #1597: FATAL ERROR: The C compiler is not able to compile a simple test program

Supermarcel10, 2024-01-01:
The following CMakeLists.txt loads perfectly fine.
```cmake
cmake_minimum_required(VERSION 3.26)

# Specify the cross-compiler
set(CMAKE_SYSTEM_NAME Generic)
set(CMAKE_SYSTEM_PROCESSOR arm)

# Path to the ARM toolchain
#set(TOOLCHAIN_DIR "C:/Program Files (x86)/Arm GNU Toolchain arm-none-eabi/13.2 Rel1/bin")
set(CMAKE_C_COMPILER arm-none-eabi-gcc.exe)
set(CMAKE_CXX_COMPILER arm-none-eabi-g++.exe)
set(CMAKE_ASM_COMPILER arm-none-eabi-gcc.exe)
set(CMAKE_DEBUGGER arm-none-eabi-gdb.exe)

# Check if SDK exists, it not then fetch it from remote git
set(PICO_SDK_FETCH_FROM_GIT ON)

# Pull in Pico SDK
include(pico_sdk_import.cmake)

project(picoTest)

# Initialise Pico SDK
pico_sdk_init()

# Include directories
include_directories(${CMAKE_SOURCE_DIR}/include)

# Source files
file(GLOB_RECURSE SOURCES "src/*.c" "src/*.cpp")

# Define the executable
add_executable(${PROJECT_NAME} ${SOURCES})

# Add pico_stdlib library which aggregates commonly used features
target_link_libraries(${PROJECT_NAME} pico_stdlib)

# Create map/bin/hex/uf2 file in addition to ELF.
pico_add_extra_outputs(${PROJECT_NAME})

```

However, when building I get the following errors:
```error
====================[ Build | picoTest | Pico ]================================
C:\Users\Marcel\AppData\Local\JetBrains\Toolbox\apps\CLion\ch-0\232.9921.42\bin\cmake\win\x64\bin\cmake.exe --build C:\Users\Marcel\CLionProjects\picoTest\cmake-build-pico --target picoTest -j 10
[1/59] Creating directories for 'ELF2UF2Build'
[2/59] Building ASM object pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/compile_time_choice.S.obj
[3/59] No download step for 'ELF2UF2Build'
[4/59] No update step for 'ELF2UF2Build'
[5/59] Linking ASM executable pico-sdk\src\rp2_common\boot_stage2\bs2_default.elf
[6/59] Generating bs2_default.bin
[7/59] No patch step for 'ELF2UF2Build'
[8/59] Generating bs2_default_padded_checksummed.S
[9/59] Performing configure step for 'ELF2UF2Build'
FAILED: elf2uf2/src/ELF2UF2Build-stamp/ELF2UF2Build-configure C:/Users/Marcel/CLionProjects/picoTest/cmake-build-pico/elf2uf2/src/ELF2UF2Build-stamp/ELF2UF2Build-configure 
cmd.exe /C "cd /D C:\Users\Marcel\CLionProjects\picoTest\cmake-build-pico\elf2uf2 && C:\Users\Marcel\AppData\Local\JetBrains\Toolbox\apps\CLion\ch-0\232.9921.42\bin\cmake\win\x64\bin\cmake.exe -DCMAKE_MAKE_PROGRAM:FILEPATH=C:/Users/Marcel/AppData/Local/JetBrains/Toolbox/apps/CLion/ch-0/232.9921.42/bin/ninja/win/x64/ninja.exe -GNinja -S C:/Users/Marcel/CLionProjects/picoTest/cmake-build-pico/_deps/pico_sdk-src/tools/elf2uf2 -B C:/Users/Marcel/CLionProjects/picoTest/cmake-build-pico/elf2uf2 && C:\Users\Marcel\AppData\Local\JetBrains\Toolbox\apps\CLion\ch-0\232.9921.42\bin\cmake\win\x64\bin\cmake.exe -E touch C:/Users/Marcel/CLionProjects/picoTest/cmake-build-pico/elf2uf2/src/ELF2UF2Build-stamp/ELF2UF2Build-configure"
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
ninja: build stopped: subcommand failed.
```

This is pretty unusual since "CMAKE_C_COMPILER" and "CMAKE_CXX_COMPILER are present in CMakeLists.txt - it looks like it's a bug/issue specifically with `pico_add_extra_outputs(${PROJECT_NAME})` which calls EL2UF2Build without passing the correct environment?

I have tried passing full directories, and PATH compiler names (as can be seen bt the commented line above the sets) in CMake. This had the same result.

To temporarily solve this I manually put "CC" and "CXX" environment variables when building, which resulted in a different error.

```error
====================[ Build | picoTest | Pico ]================================
C:\Users\Marcel\AppData\Local\JetBrains\Toolbox\apps\CLion\ch-0\232.9921.42\bin\cmake\win\x64\bin\cmake.exe --build C:\Users\Marcel\CLionProjects\picoTest\cmake-build-pico --target picoTest -j 10
[1/51] Performing configure step for 'ELF2UF2Build'
FAILED: elf2uf2/src/ELF2UF2Build-stamp/ELF2UF2Build-configure C:/Users/Marcel/CLionProjects/picoTest/cmake-build-pico/elf2uf2/src/ELF2UF2Build-stamp/ELF2UF2Build-configure 
cmd.exe /C "cd /D C:\Users\Marcel\CLionProjects\picoTest\cmake-build-pico\elf2uf2 && C:\Users\Marcel\AppData\Local\JetBrains\Toolbox\apps\CLion\ch-0\232.9921.42\bin\cmake\win\x64\bin\cmake.exe -DCMAKE_MAKE_PROGRAM:FILEPATH=C:/Users/Marcel/AppData/Local/JetBrains/Toolbox/apps/CLion/ch-0/232.9921.42/bin/ninja/win/x64/ninja.exe -GNinja -S C:/Users/Marcel/CLionProjects/picoTest/cmake-build-pico/_deps/pico_sdk-src/tools/elf2uf2 -B C:/Users/Marcel/CLionProjects/picoTest/cmake-build-pico/elf2uf2 && C:\Users\Marcel\AppData\Local\JetBrains\Toolbox\apps\CLion\ch-0\232.9921.42\bin\cmake\win\x64\bin\cmake.exe -E touch C:/Users/Marcel/CLionProjects/picoTest/cmake-build-pico/elf2uf2/src/ELF2UF2Build-stamp/ELF2UF2Build-configure"
-- The C compiler identification is GNU 13.2.1
-- The CXX compiler identification is GNU 13.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: C:/Program Files (x86)/Arm GNU Toolchain arm-none-eabi/13.2 Rel1/bin/arm-none-eabi-gcc.exe
-- Check for working C compiler: C:/Program Files (x86)/Arm GNU Toolchain arm-none-eabi/13.2 Rel1/bin/arm-none-eabi-gcc.exe - broken
CMake Error at C:/Users/Marcel/AppData/Local/JetBrains/Toolbox/apps/CLion/ch-0/232.9921.42/bin/cmake/win/x64/share/cmake-3.26/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "C:/Program Files (x86)/Arm GNU Toolchain arm-none-eabi/13.2 Rel1/bin/arm-none-eabi-gcc.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: C:/Users/Marcel/CLionProjects/picoTest/cmake-build-pico/elf2uf2/CMakeFiles/CMakeScratch/TryCompile-u97dhs
    
    Run Build Command(s):C:/Users/Marcel/AppData/Local/JetBrains/Toolbox/apps/CLion/ch-0/232.9921.42/bin/ninja/win/x64/ninja.exe -v cmTC_c2585 && [1/2] "C:\PROGRA~2\Arm GNU Toolchain arm-none-eabi\13.2 Rel1\bin\arm-none-eabi-gcc.exe"    -o CMakeFiles/cmTC_c2585.dir/testCCompiler.c.obj -c C:/Users/Marcel/CLionProjects/picoTest/cmake-build-pico/elf2uf2/CMakeFiles/CMakeScratch/TryCompile-u97dhs/testCCompiler.c
    [2/2] cmd.exe /C "cd . && "C:\PROGRA~2\Arm GNU Toolchain arm-none-eabi\13.2 Rel1\bin\arm-none-eabi-gcc.exe"   CMakeFiles/cmTC_c2585.dir/testCCompiler.c.obj -o cmTC_c2585.exe -Wl,--out-implib,libcmTC_c2585.dll.a -Wl,--major-image-version,0,--minor-image-version,0  -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32 && cd ."
    FAILED: cmTC_c2585.exe 
    cmd.exe /C "cd . && "C:\PROGRA~2\Arm GNU Toolchain arm-none-eabi\13.2 Rel1\bin\arm-none-eabi-gcc.exe"   CMakeFiles/cmTC_c2585.dir/testCCompiler.c.obj -o cmTC_c2585.exe -Wl,--out-implib,libcmTC_c2585.dll.a -Wl,--major-image-version,0,--minor-image-version,0  -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32 && cd ."
    C:/PROGRA~2/Arm GNU Toolchain arm-none-eabi/13.2 Rel1/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld.exe: unrecognized option '--major-image-version'
    C:/PROGRA~2/Arm GNU Toolchain arm-none-eabi/13.2 Rel1/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld.exe: use the --help option for usage information
    collect2.exe: error: ld returned 1 exit status
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

-- Configuring incomplete, errors occurred!
ninja: build stopped: subcommand failed.
```

I have tried two ARM toolchain versions and they both fail the same way. Toolchain version 13.2 Rel1 is the replacement of 10.3.1 (as shown in RaspberryPi tutorials) which is deprecated.

I have also tried pulling in pico_examples and it fails in the same manner, which makes me believe it's an error with pico-sdk rather than the compilers or configuration.

If anyone has any suggestions of what I can try, feel free to comment and I'll be able to test further.

Host OS: Windows 10 22H2 x86
Target OS: Generic RP2040 ARM (RPi Pico W)
IDE: CLion CL-232.9921.42 September 12, 2023
ARM Toolchains: GNU 10.3.1, 13.2 Rel1

lurch, 2024-01-01:
There are actually two compilers needed during building an RP2040 UF2 file. A native (x64) compiler used to build the native `elf2uf2.exe` executable, and then the (arm) cross-compiler used to build `picoTest.elf` (which `elf2uf2.exe` then turns into `picoTest.uf2`). And it looks like at least one of your errors is caused by trying to use the cross-compiler to build the native elf2uf2 executable.

I've personally not used CLion, but perhaps section 10.2 of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf will steer you in the right direction?

Supermarcel10, 2024-01-04:
I see. I'll take a look at section 10.2 and see if I can start from scratch and get it working by any means.

raarts, 2024-01-05:
I had the same error on CLion, but I can't remember if the circumstances were the same. Anyway, I **do** remember the solution: In Build, Execution, Deployment -> CMake I had the toolchain set to gnu_arm_embedded. Reverting it back to Use Default fixed the problem. Maybe this works for you.

Supermarcel10, 2024-01-08:
Ok I see! That has resolved my issue with being able to build. I cannot express how long I spent on this, thinking I'm doing something wrong with the other compiler. Didn't realise I'm not supposed to use the ARM Toolchain compiler for the entirity of the build. Thank you!

closed 2024-01-08 by Supermarcel10

bactone, 2024-03-29:
@raarts however, I use the default ,but same error :
[image]
[image]
[image]
it really drive me crazy , even though I have re-install Clion, but same error, which does not happen before

raarts, 2024-03-29:
Maybe it has to do with the fact I use a Mac M1?