raspberrypi/pico-sdk #1598: panic("No program space") on second call of pio_add_program

berndbenner, 2024-01-01:
I am searching for a strange panic-error on loading the second pio program.
( pico-sdk version 1.5.1 ).

I have a panic("No program space").

# What happens:

The first pio program to load has a length of 13.
pio_find_offset_for_program(pio, program) returns 19 
pio_add_program_at_offset will set:
uint32_t program_mask = (1u << program->length) - 1;
This is 0x1FFF (8191)    

your bitvector array marker _used_instruction_space has only 32 bit per pio.
_used_instruction_space[pio_get_index(pio)] |= program_mask << offset;
will be set to 0xFFF80000.

The second program to load has a length of 10.
so 
uint32_t program_mask = (1u << program->length) - 1;
will be set to 0x3FF ( 1023)

and pio_find_offset_for_program

```C 
// work down from the top always
 for (int i = 32 - program->length; i >= 0; i--) {
    if (!(used_mask & (program_mask << (uint) i))) {
              return i;
   }
 }
```
will not find a free offset address.

It seems pio_find_offset_for_program with the "work down from the top" code is a little untested with more than one program loads!
(4 x max 15 bit mask) will not fit in a 32  bit vector per pio. 

Please change all allocation masks to 64 bit or use another partitioning algorythm.

I will send you my source code changes. 
I have changed all allocation masks to 64 bit and it works.
[pio_fixes.c.txt](https://github.com/raspberrypi/pico-sdk/files/13805120/pio_fixes.c.txt)

lurch, 2024-01-01:
> your bitvector array marker _used_instruction_space has only 32 bit per pio.

That's because each PIO only has a 32-word instruction memory, so it uses 32-bits to keep track of which of the 32 instruction-slots have been used already.

Doing some back-of-the-envelope calculations, if your first 13-instruction program gets loaded at offset 19 (i.e. it's using instruction-slots 19 to 31 inclusive), shouldn't your second 10-instruction program get loaded at offset 9 (i.e. using instruction-slots 9 to 18 inclusive) ?
( `0xFFF80000 & (0x3FF << 9)` equals zero )

> (4 x max 15 bit mask) will not fit in a 32 bit vector per pio.

Huh?

berndbenner, 2024-01-01:
Ok Thank you! I have missunderstand the docs. So 4 state machines has a maximum of 32 instructions in sum!

closed 2024-01-01 by berndbenner