raspberrypi/pico-sdk #1601: pio_gpio_init will kill output of sm on other pio

berndbenner, 2024-01-06:
I am using a generator sm on a pio to generate output pulses. On other slave-sm(s) I am reading back the output of the generator sm and wait for pulse edges with wait 0 ...,wait 1 ...-instructions to synchronise the output to the generator sm. This works very well, if the slave sms are on the same pio.  

If I try do run the slave-sm on the other pio.  The call of pio_gpio_init of the read back output on the other pio will kill the output of the generator sm on the generator pio.

It seems that the pio_gpio_init-call will (re)initialize the pin-direction of the output pin of the other state machine to input.

The only solution I found was to read back the output on a bridged input pin.
for the slave-sm on the other pio.

Please describe this behavour in your documentation.  

My understanding of "PIO controls the output level and direction of up to 32 GPIOs, and can observe their input levels. ... ranges can overlap" 
is different.

lurch, 2024-01-10:
> It seems that the pio_gpio_init-call will (re)initialize the pin-direction of the output pin of the other state machine to input.

https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#gafa244b1be8f53a329db9d26298e054bb says "Note that this is not necessary for a state machine to be able to read the input value from a GPIO, but only for it to set the output value or output enable.", which I guess implies that you shouldn't be re-initing the GPIO on your "slave" PIO?

peterharperuk, 2024-02-14:
It sounds like @lurch has identified your issue?

closed 2024-02-14 by peterharperuk