raspberrypi/pico-sdk #1602: Panic on Core 0 When Initializing MQTT Client with FreeRTOS SMP

DarkNebula0, 2024-01-07:
I am experiencing a system panic on Core 0 of the RP2040 when attempting to initialize the MQTT client using FreeRTOS SMP. The issue arises specifically during the MQTT client initialization process from the lwIP library.

**Environment:**
Hardware: pico_w
OS: FreeRTOS SMP
Library: lwIP <lwip/apps/mqtt.h>

**Linked Libraries**
```cmake
target_link_libraries(test
        pico_cyw43_arch_lwip_sys_freertos
        pico_stdlib
        pico_lwip_iperf
        FreeRTOS-Kernel-Heap4
        pico_lwip_mbedtls
        pico_mbedtls
        pico_lwip_mqtt
)
```

**Stack Trace:**
```
__breakpoint platform.h:345
_exit runtime.c:187
panic runtime.c:325
sys_mutex_lock sys_arch.c:218
mem_malloc mem.c:859
mem_calloc mem.c:1010
mqtt_client_new mqtt.c:1264 <-------------------
CNetwork::CNetwork Network.h:22
CSingleton::instance Singleton.h:16
__static_initialization_and_destruction_0 Network.cpp:15
_GLOBAL__sub_I_NetworkInstance Network.cpp:150
runtime_init runtime.c:177
platform_entry crt0.S:258
```

**Detailed Description:**
The panic occurs when I call mqtt_client_new() from `lwip/apps/mqtt.h` This function eventually calls mem_malloc from pico-`sdk\lib\lwip\src\core\mem.c`, which in turn calls sys_mutex_lock. The failure happens with an LWIP_ASSERT, specifically `LWIP_ASSERT("mutex->mut != NULL", mutex->mut != NULL);`.

All actions leading to the panic are executed on core 0, which suggests an issue with mutex handling in a multi-core environment or a misconfiguration.

**Steps to Reproduce:**
Initialize the RP2040 with FreeRTOS SMP.
Call mqtt_client_new() to initialize the MQTT client.
Observe the panic on core 0 during the execution of sys_mutex_lock.

**FreeRTOSConfig.h**
```c++
/*
 * FreeRTOS V202212.00
 * Copyright (C) 2020 Amazon.com, Inc. or its affiliates.  All Rights Reserved.
 *
 * Permission is hereby granted, free of charge, to any person obtaining a copy of
 * this software and associated documentation files (the "Software"), to deal in
 * the Software without restriction, including without limitation the rights to
 * use, copy, modify, merge, publish, distribute, sublicense, and/or sell copies of
 * the Software, and to permit persons to whom the Software is furnished to do so,
 * subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in all
 * copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS
 * FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR
 * COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER
 * IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN
 * CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
 *
[... 208 lines cut ...]
#define TCPIP_DEBUG                 LWIP_DBG_OFF
#define PPP_DEBUG                   LWIP_DBG_OFF
#define SLIP_DEBUG                  LWIP_DBG_OFF
#define DHCP_DEBUG                  LWIP_DBG_OFF

#if !NO_SYS
#define TCPIP_THREAD_STACKSIZE 1024
#define DEFAULT_THREAD_STACKSIZE 1024
#define DEFAULT_RAW_RECVMBOX_SIZE 8
#define TCPIP_MBOX_SIZE 8
#define LWIP_TIMEVAL_PRIVATE 0

// not necessary, can be done either way
#define LWIP_TCPIP_CORE_LOCKING_INPUT 1

// ping_thread sets socket receive timeout, so enable this feature
#define LWIP_SO_RCVTIMEO 1
#endif

#define LWIP_ALTCP               1
#define LWIP_ALTCP_TLS           1
#define LWIP_ALTCP_TLS_MBEDTLS   1
#define MEMP_NUM_SYS_TIMEOUT     LWIP_NUM_SYS_TIMEOUT_INTERNAL + 1
#endif
```

dpslwk, 2024-01-09:
do you have the call to `mqtt_*` wrapped between  `cyw43_arch_lwip_begin()` and `cyw43_arch_lwip_end()` ?

DarkNebula0, 2024-01-10:
> cyw43_arch_lwip_begin

Sorry I was on the road and couldn't test it earlier. 

Just tested it, when I call cyw43_arch_lwip_begin I end up with an `isr_hardfault`

```
isr_hardfault crt0.S:98
<signal handler called> 0x00000000fffffff9
vtable for __gnu_cxx::stdio_sync_filebuf<char, std::char_traits<char> > 0x00000000100dffc0
async_context_acquire_lock_blocking async_context.h:207
cyw43_thread_enter cyw43_driver.c:153
cyw43_arch_lwip_begin cyw43_arch.h:274
CNetwork::CNetwork Network.h:24
CSingleton::instance Singleton.h:16
__static_initialization_and_destruction_0 Network.cpp:15
_GLOBAL__sub_I_NetworkInstance Network.cpp:150
runtime_init runtime.c:177
platform_entry crt0.S:258
```

DarkNebula0, 2024-01-10:
> Have you initialise lwip? Have you initialise cyw43?

´lwip_init´ did the trick.

closed 2024-01-10 by DarkNebula0