raspberrypi/pico-sdk #1605: newlib not included in compile_commands.json for some files

ast, 2024-01-12:
I generated a cpp project named keyboard with pico-project-generator.

Using cmake 3.27.4 with

`set(CMAKE_EXPORT_COMPILE_COMMANDS ON)`

, to generate a compile_commands.json for use with clangd (in my case with eglot on emacs).

For the file `keyboard.cpp -I/usr/include/newlib is missing, but it's there for other files such is irq_handler_chain.S and most if not all other .S files.

This causes code completion not to work correctly.

I have been unable to find the cause of this, but worked around it using:

`set(CMAKE_CXX_STANDARD_INCLUDE_DIRECTORIES /usr/include/newlib)`

peterharperuk, 2024-01-12:
Why is your pico project using /usr/include/newlib?

vulpes2, 2024-02-08:
Just ran into this, on a Fedora system you can set this to make clangd work:
```
set(CMAKE_C_STANDARD_INCLUDE_DIRECTORIES /usr/arm-none-eabi/include)
set(CMAKE_CXX_STANDARD_INCLUDE_DIRECTORIES /usr/arm-none-eabi/include)
```

will-v-pi, 2024-08-09:
There have been some changes to the way newlib is included in SDK 2.0.0, so could you try with the new SDK and see if this is still an issue?

vulpes2, 2024-08-10:
Nothing has changed so far. Without `set(CMAKE_C_STANDARD_INCLUDE_DIRECTORIES /usr/arm-none-eabi/include)` things like `stdio.h` still cannot be found. Tested with the latest version of `pico-sdk` and `pico-examples`.

will-v-pi, 2024-08-21:
Looking at the clangd docs, the best way to get it working seems to be to specify `--query-driver` on the command line with the path to your compiler, rather than specifying system include paths manually (https://clangd.llvm.org/troubleshooting#cant-find-standard-library-headers-map-stdioh-etc)

This works with pico examples, by generating compile_commands.json, and then running
```
clangd --query-driver=/usr/bin/* --check=hello_world/usb/hello_usb.c
```
for example - it now picks up the newlib headers correctly. These can be added 

The `--query-driver` argument is documented [here](https://clangd.llvm.org/guides/system-headers#query-driver)

will-v-pi, 2024-11-12:
Closing this, as I don't think the newlib libraries should be included in `compile_commands.json` - instead the intellisense tool used (eg `clangd`) should query the compiler for the system library locations

closed 2024-11-12 by will-v-pi