raspberrypi/pico-sdk #1606: The timer doesn't work when using FreeRTOS in a multicore environment.

hajmedwissem, 2024-01-12:
I use FreeRTOS on the Raspberry Pi Pico to utilize both cores. In one of my FreeRTOS tasks running on core 1, I have created a timer with a callback function scheduled to run every 90 microseconds. However, the function is never called. I use the following code to create the timer and set the alarm:

add_alarm_in_us(90, timer_callback, NULL, true);

Could anyone explain why the timer isn't starting? 
Thank you, guys.

peterharperuk, 2024-01-12:
Is the alarm_id returned <0?

hajmedwissem, 2024-01-12:
After debugging, the value of alarm_id returned by the add_alarm_in_us() function is equal to 1

kilograham, 2024-05-18:
please provide a more complete repro scenario (or ask on the forums https://forums.raspberrypi.com/viewforum.php?f=143)

closed 2024-05-18 by kilograham