raspberrypi/pico-sdk #1607: Pico Blink vs Pico Blink W

mischievous, 2024-01-12:
Am I really doing this correctly, the pico blink is 20K, and the pico w blink is 260K?  Is there anything that I can reduce from the pico w image?  

Here is the important parts of the CMakeLists.txt.  Blink.c is either the standard blink.c or the wifi example for blink.c.  Then I turn on/off the pico_cyw43_arch_none as needed.  

pico_sdk_init()

#
add_executable(${TARGET}
  blink.c
)

target_include_directories(${TARGET} PRIVATE
    ${CMAKE_CURRENT_LIST_DIR}
)

target_link_libraries(${TARGET}
        pico_stdlib              # for core functionality
        pico_cyw43_arch_none     # we need Wifi to access the GPIO, but we don't need anything else
        )

pico_add_extra_outputs(${TARGET})

peterharperuk, 2024-01-12:
It's the WiFi firmware

mischievous, 2024-01-13:
So, the wifi firmware in pico_cyw43_arch_none (which should exclude) the wifi part is using an additional 240k to just blink a led.  Ouch!

Might have to re-think what I am trying to do here... Maybe use a pico as the main part and a pico-w as the network I/F connected via a pseudo usb thing.

peterharperuk, 2024-01-13:
There are lots of gpios available if you want to connect a led. The none in pico_cyw43_arch_none refers to their being no lwip IP attack

closed 2024-01-13 by peterharperuk

mischievous, 2024-01-13:
Thanks for your help

lurch, 2024-01-14:
See also https://github.com/raspberrypi/pico-examples/issues/421 and https://github.com/raspberrypi/pico-feedback/issues/364

aallan, 2024-01-15:
> Am I really doing this correctly, the pico blink is 20K, and the pico w blink is 260K?

The onboard LED on Pico W is connected to the 43439 WiFi chip rather than the RP2040. You're loading the WiFi firmware to talk to the 43439.