raspberrypi/pico-sdk #1611: SWD failures seemingly related to openocd utility

seamusdemora, 2024-01-22:
On a RPi 5 running bookworm Lite; "fairly close" to pristine.

I have installed iaw the instructions; i.e. ran  `./pico_setup.sh`
I have wired up the SWD iaw the documentation; i.e.
```
I've triple-checked the wiring, and it is IAW the Guide:
SWDIO (Pico) ===> connected to ===> GPIO 24 (RPi pin 18)
SWCLK (Pico) ===> connected to ===> GPIO 25 (RPi pin 22)
GND (Pico) ===> connected to ===> GND (RPi pin 20)
```

Unfortunately, it does not work: 
```
openocd -f interface/raspberrypi-swd.cfg -f target/rp2040.cfg -c "program blink/blink.elf verify reset exit"
Open On-Chip Debugger 0.12.0-g4d87f6d (2024-01-20-06:51)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
WARNING: Unknown type of the host SoC. Expect JTAG/SWD clock slower than requested.
Warn : TMS/SWDIO moved to GPIO 8 (pin 24). Check the wiring please!
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Error: open: Permission denied
** OpenOCD init failed **
shutdown command invoked
```

I emphasize again that the wiring is done IAW the documentation. So given that the error message (or the documentation) is incorrect, what must be done to get this working?

aallan, 2024-01-22:
Can you point me at the documentation (website, or PDF) you are following?

seamusdemora, 2024-01-22:
Surely, it's the [Getting Started Guide](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf).

aallan, 2024-01-22:
These instructions need updating for Raspberry Pi 5. See raspberrypi/openocd#93 the PR linked from there raspberrypi/pico-setup#28. Now folks are back from the holidays it's likely we'll see some movement there.

aallan, 2024-01-22:
@nathan-contino Can you take a look at raspberrypi/openocd#93 and see if we can squeeze this into the upcoming documentation release? Willing to punt for a couple of days if you think it's doable. Thanks!

seamusdemora, 2024-01-22:
@aallan :  

Thanks for the rapid update!  

After reading [raspberrypi/openocd#93](https://github.com/raspberrypi/openocd/issues/93) I am still a little unclear on the fix... is a change to the config file all that's required, or is there some code in linuxgpiod.c that must also be updated?

cleverca22, 2024-01-23:
@seamusdemora just create the `test.cfg` file and use it as shown in the comments
shouldnt be any need to change source in `linuxgpiod.c`

seamusdemora, 2024-01-23:
@cleverca22 : 

Do you mean this file?: 

```
clever@raspberrypi:~ $ cat test.cfg
adapter driver linuxgpiod
adapter gpio swclk 20 -chip 4
adapter gpio swdio 21 -chip 4
``` 

And apologies, but I'm not following your **context**...  I mean where does this test.cfg file go? It looks like you're running `openocd` from your `~/` directory, but where's `target`? 

P.S. I am not at all familiar with `openocd`.

aallan, 2024-01-26:
Fixed in the next documentation update, which should be out within the next few days.

closed 2024-01-26 by aallan

cleverca22, 2024-02-02:
@seamusdemora 
> but where's `target`?

any time openocd cant find a file you gave it, then it will search its built in scripts
which are somewhere under /usr/local if you built from source

https://github.com/raspberrypi/openocd/tree/rp2040-v0.12.0/tcl/target
this is the source that those built-in scripts come from

hasenradball, 2024-02-11:
@aallan 
Hi Sir, my I ask a question?
Is it meaning when the topic is closed now does it mean, that the Pico C SDK runs fine on the pi 5 now?
Or there are still open issues?

Best regards
Frank

hasenradball, 2024-02-15:
@nathan-contino 
I checked the new documentation, but I don't get it to work.

Are the SWDIO and SWCLK Ports changed in Pi5?

I created a file like in the description mentioned and it look like this:

[image]

And the error meassage looks like this:
[image]

What do I wrong???

hasenradball, 2024-02-15:
@nathan-contino
If I set the cfg to this:

```
source [find interface/raspberrypi-native.cfg]

adapter driver linuxgpiod

adapter gpio swclk 20 -chip 4
adapter gpio swdio 21 -chip 4

adapter speed 5000
```

I get this error:
[image]

nathan-contino, 2024-02-15:
@hasenradball sorry about that. It looks like we have more investigating to do.

Can you try compiling with the options in https://github.com/raspberrypi/pico-setup/pull/28 ? That's the only reason I can think of that this would fail with the current setup. Otherwise I'll have to do more digging and reproduce the issue.

hasenradball, 2024-02-15:
@nathan-contino 
Can you please give me the options I should use, Thanks

nathan-contino, 2024-02-15:
You should be able to copy and paste the contents of [`pico_setup.sh`](https://github.com/lurk101/pico-setup/blob/master/pico_setup.sh) from the fork used in that PR into a bash file on your own system, then run it locally to install OpenOCD with pi 5-friendly GPIO libraries.

hasenradball, 2024-02-15:
@nathan-contino
If you mean this???

This PR is still open!

[image]

hasenradball, 2024-02-15:
@nathan-contino 
Running the `pico_setup.sh` with the above mentioned options, does not help.

Can ypu please paste how the cfg-file for th Pi5 should look like?

Thank you.

hasenradball, 2024-02-15:
> You should be able to copy and paste the contents of [`pico_setup.sh`](https://github.com/lurk101/pico-setup/blob/master/pico_setup.sh) from the fork used in that PR into a bash file on your own system, then run it locally to install OpenOCD with pi 5-friendly GPIO libraries.
> 
> It seems that the version of openocd you're using wasn't built with those libraries, hence your issue.
> 
> I tested on Linux and macOS, but I don't have a Windows box to test on and I can't cover all possible environments anyway.

I tested it on the Pi5 with rasparian bookworm.
No changes in the error message.

Can you give a screenshot of your pi5 cfg file for openocd?

nathan-contino, 2024-02-15:
Thank you for your feedback. I'll investigate this when I get some spare cycles to try to reproduce the issue.

hasenradball, 2024-02-15:
> Thank you for your feedback. I'll investigate this when I get some spare cycles to try to reproduce the issue.

Can you please paste the content of your config!!!
To be able to check how it should look like.

ivt18, 2024-02-15:
Good evening, I am currently facing similar issues with this. After building OpenOCD from source, and with `libgpio-dev`, I now get the following output:
```
$ openocd -f raspberrypi5-swd.cfg -f target/rp2040.cfg -c "program build/src/main.elf verify reset exit"
Open On-Chip Debugger 0.12.0-g4d87f6d (2024-02-15-22:50)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Warn : An adapter speed is not selected in the init scripts. OpenOCD will try to run the adapter at the low speed (100 kHz)
Warn : To remove this warnings and achieve reasonable communication speed with the target, set "adapter speed" or "jtag_rclk" in the init scripts.
Info : Linux GPIOD JTAG/SWD bitbang driver
Info : This adapter doesn't support configurable speed
Error: Failed to connect multidrop rp2040.dap0
in procedure 'program'
** OpenOCD init failed **
shutdown command invoked
```
I have connected the RPi 5 SWD according to the [Getting Started Guide](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf), and I'm using the following config:
```
$ cat raspberrypi5-swd.cfg 
adapter driver linuxgpiod
adapter gpio swclk 20 -chip 4
adapter gpio swdio 21 -chip 4
```
Am I doing something wrong?

hasenradball, 2024-02-15:
@nathan-contino @aallan @lurch 
What the hell was done in the review before the Documentation update release???

Hi I don' t belive what I see here....
Have you read the last sentence in this screenshot?

[image]

Somebody changed the PINs for `SWDIO` and `SWCLK`, because it was for him a better place to use these pins but this does nothing have to do with the `Getting startetd Guide`
And some stupid guy takes this over into an official Documentation release???

If I completly rebuild openocd like this approach:
[image]

 

The working config file for the official Pi Documentation should look like this:

```
adapter driver linuxgpiod
adapter gpio swclk 25 -chip 4
adapter gpio swdio 24 -chip 4
```
And PLEASE !!!!

1) Can anyone merge this PR immediately into the master [PR28](https://github.com/raspberrypi/pico-setup/pull/28)
AND 
2) The guy how has tomatos on the eyes please correct to official release for this in chapter 5:
[Getting Started Guide](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf)

[image]

ivt18, 2024-02-15:
@hasenradball Welp, that fixed it. I must say I was skeptical at the 21/20 pins indicated on the documentation, but trusted it would be correct. I stand corrected.

nathan-contino, 2024-02-15:
Thanks for all of the feedback here. I've updated the source code and we'll roll out the fix in the next release. Sorry that the config file pin assignment was wrong in that last release.

Please try and be polite; mistakes do happen in documentation. Fortunately I have successfully removed the tomatoes from my eyes so this one shouldn't happen again.

hasenradball, 2024-02-15:
Thanks for your feedback.
I was also suspicious but, I believed an actual official release of documentation more then my doupt.

hasenradball, 2024-02-15:
> Thanks for all of the feedback here. I've updated the source code and we'll roll out the fix in the next release. Sorry that the config file pin assignment was wrong in that last release.
>
> Please [...]

@nathan-contino 
Sorry I was so so angry...

hasenradball, 2024-02-15:
@nathan-contino 
If you are interested in, I can give you an example to setup a task in vscode to flash the code on the pico via key short cut.

hasenradball, 2024-02-16:
@nathan-contino Hi just for clarifikation,
you also have to take over the changes from the PR28 (the changes in the pico_setup.sh.

I would recommend to take over this PR immediately, because rebuild of openocd is not so easy than using the setup script.
And running the setup script twice after changed the parameters won‘ t help here.

lurk101, 2024-02-18:
I'm afraid PR#28 is now closed. I had to delete the originating repo. I figured over a month with no merge was long enough to be ignored!

hasenradball, 2024-02-18:
> I'm afraid PR#28 is now closed. I had to delete the originating repo. I figured over a month with no merge was long enough to be ignored!

Welp, that seems to be a main problem I faced a similar problem.
:-(

hasenradball, 2024-02-18:
@nathan-contino 
One Question, you mentioned you took over the changes.
Did you took over both parts?
a) the change in the pico_ setup.sh
and
b) the change in the docu?

a) could be take over immidiately, from my undastaning.

lurk101, 2024-02-18:
> > I'm afraid PR#28 is now closed. I had to delete the originating repo. I figured over a month with no merge was long enough to be ignored!
> 
> Welp, that seems to be a main problem I faced a similar problem. :-(

I think you can still use it. Just select closed PRs instead of open ones, then you can look at the commit.

hasenradball, 2024-02-20:
> I think you can still use it. Just select closed PRs instead of open ones, then you can look at the commit.

Could you please reopen it?

lurk101, 2024-02-20:
Not possible. The originating repo has been deleted.