raspberrypi/pico-sdk #1615: No RPi5 config file for openocd?

seamusdemora, 2024-01-25:
I'm doing development work on a Pico using a RPi5 as the development machine. I'm also using the "Debug Probe" thing-y that you sell. Things are moving along OK, but I'm not doing anything very challenging at the moment. 

_**However**_ there does not appear to be a config file that applies to the RPi5 in the `openocd` folder. I draw this conclusion from the fact that each time I invoke `openocd` without the `-c "adapter speed 5000"` option in the command line, I get a warning in the output: 

```bash
$ sudo openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "program ina260_trial.elf verify reset exit"
Open On-Chip Debugger 0.12.0-g4d87f6d (2024-01-22-05:04)
...
Warn : An adapter speed is not selected in the init scripts. OpenOCD will try to run the adapter at the low speed (100 kHz)
Warn : To remove this warnings and achieve reasonable communication speed with the target, set "adapter speed" or "jtag_rclk" in the init scripts.
...
```
Note the **Warning** suggests setting `adapter speed` _"in the init scripts"_. I don't know what the heck they mean by _"init scripts"_, but I could find nothing appropriate in _any_ script in `/pico/openocd` for setting `adapter speed`. So, I tried `grep`-ing in `pico/openocd`:
```bash
$ grep -rni "adapter speed" | less
```
This produced a rather extensive list, mostly in the `pico/openocd/tcl` folder. After some searching, I found a few candidates:
```
tcl/board/pico-debug.cfg
tcl/interface/raspberrypi-swd.cfg
tcl/interface/raspberrypi-native.cfg
tcl/interface/raspberrypi-native.cfg
tcl/target/bcm2835.cfg
testing/test-bcm2835gpio-deprecated-commands.cfg
tcl/target/bcm2711.cfg
```
However, after inspecting each of these files, it seems that none of them are appropriate for the RPi5 and appear to be a logical place for setting `adapter speed`. Also, there is no **cfg** file for **bcm2712**.

If I've missed something, please let me know. If I haven't, I would surely appreciate a configuration file that sets `adapter speed`, and otherwise covers the RPi5 in `openocd`.

kilograham, 2024-01-25:
see #1611

seamusdemora, 2024-01-25:
@kilograham 
Hey Graham - yeah, I'm aware of #1611 as I submitted that one also. However, #1611 was submitted for a different issue - an issue I've (at least temporarily) overcome by using the "Debug Probe". ... or am I missing something obvious?

peterharperuk, 2024-01-25:
I use a debug probe with pi5 every day. I'll check if I get this warning

peterharperuk, 2024-01-26:
I seem to have always been running openocd like this which avoids the issue...

openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "adapter speed 5000"

P33M, 2024-01-26:
The realisable swclk depends on the hardware in question. It's explicitly a user choice to select a clock compatible with their hardware. 5MHz is a recommendation, but should not be a default.

peterharperuk, 2024-01-26:
Is 100 kHz a sensible default? That's what your get if you don't set it. I've heard reports of pico debugging being slow - I bet this is the reason

peterharperuk, 2024-01-26:
Here would be more appropriate? https://github.com/openocd-org/openocd

lurch, 2024-01-26:
Our own openocd repo is still what https://github.com/raspberrypi/pico-setup/blob/master/pico_setup.sh#L135 points to, so does that need updating too? :man_shrugging:  (I'm still not 100% familiar with how SWD / OpenOCD / PicoProbe / GDB and their various upstream / downstream forks all connect together)

peterharperuk, 2024-01-26:
> so does that need updating too?

I would guess, "yes"

P33M, 2024-01-26:
I'm not averse to wrapping interface/cmsis-dap.cfg in interface/debugprobe.cfg and setting the speed there - it's a reasonable assumption that the JST connector + ribbon will be in use, which will easily do 5MHz.

seamusdemora, 2024-01-30:
> The realisable swclk depends on the hardware in question. It's explicitly a user choice to select a clock compatible with their hardware. 5MHz is a recommendation, but should not be a default.

@P33M : 

I mentioned in the first sentence of this issue that I was using the Debug Probe. I assume that settles the question re your comment re the "hardware in question"?? 

And if one were using a couple of GPIO from the RPi (instead of the Probe), could they not match the speed of the Debug Probe?

seamusdemora, 2024-01-30:
I'm going to "muddy the water" a bit with some information I discovered earlier today. If someone thinks this needs go in a separate issue, pls let me know & I'll move it. 

In addition to the missing `adapter speed` setting, it has also bothered me that I'm required to use `sudo` to run `openocd`. This gets a bit complex, so I've separated into two (2) threads to hopefully make it less so.

### 1. To which config file should the `adapter speed 5000` be added?
I **assumed** that all of the required files were installed under the `/home/pi/pico` folder. The command I use to run `openocd` is as follows:
```bash
$ sudo openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "program ina260_trial.elf verify reset exit"
```
I assumed that the two files called out in this command were (somewhere) under `/home/pi/pico`.   I further assumed that `rp2040.cfg` was the most likely one to add the `adapter speed`.  I used `locate` to find rp2040.cfg. Turns out there are two copies of that file (at least on my system):
```bash
$ locate rp2040.cfg
/home/pi/pico/openocd/tcl/target/rp2040.cfg
/usr/local/share/openocd/scripts/target/rp2040.cfg
``` 
And so I edited the file `/home/pi/pico/openocd/tcl/target/rp2040.cfg` to add the single line: `adapter speed 5000`. When I tried running `openocd` again, it still defaulted to 100 kHz. And so then I tried the same edit in `/usr/local/share/openocd/scripts/target/rp2040.cfg`. **It worked... this "did the trick".** 

I feel like I've made progress, but wonder the point of putting two copies of rp2040.cfg on the system, and then using the one in `/usr/local/share` which is owned by `root`? 

### 2. Why is `sudo` required to run `openocd`? 
As all the files in `/usr/local/share/openocd` are owned by `root` I thought this might have something to do with it... perhaps `openocd` wrote something to `rp2040`??? Anyway, I thought changing it was worth a try: 
```bash
$ sudo chown pi:pi /usr/local/share/openocd/scripts/target/rp2040.cfg
```
But this made no difference. There are also duplicates of the `cmsis-dap.cfg` file used in the `openocd` command, and so I `chown`d that file also:
```bash
$ sudo chown pi:pi /usr/local/share/openocd/scripts/interface/cmsis-dap.cfg
```
`openocd` still errors out without `sudo`: 

```bash
$ openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "program ina260_trial.elf verify reset exit"
Open On-Chip Debugger 0.12.0-g4d87f6d (2024-01-22-05:04)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Error: unable to find a matching CMSIS-DAP device
** OpenOCD init failed **
shutdown command invoked
```
At least the error message seems to pin the fault on `CMSIS-DAP`. 

Any ideas what's going on here, and how to fix it so that `sudo` is not required?  Or is that even possible? I think I've run out of steam at present because I don't really understand what `openocd` does. Hoping someone who's more knowledgeable has a solution.

lurch, 2024-01-30:
> I feel like I've made progress, but wonder the point of putting two copies of rp2040.cfg on the system, and then using the one in `/usr/local/share` which is owned by `root`?

I assume you've used the `pico_setup.sh` script? The `/home/pi/pico/openocd/tcl/target/rp2040.cfg` file is created by the "[git clone](https://github.com/raspberrypi/pico-setup/blob/master/pico_setup.sh#L135)", and the `/usr/local/share/openocd/scripts/target/rp2040.cfg` file is created by the "[make install](https://github.com/raspberrypi/pico-setup/blob/master/pico_setup.sh#L140)". (This is the way that things are typically done on Linux when installing software from sourcecode, and isn't specific to openocd). If you're really concerned about there being "two copies of the file", you could delete the entire `/home/pi/pico/openocd` folder after the `pico_setup.sh` script has finished, if you want.

> Any ideas what's going on here, and how to fix it so that `sudo` is not required?

I suspect that openocd needs root privileges to be able to talk to the hardware (i.e. the hardware that `interface/cmsis-dap.cfg` is specifying)? That's probably fixable with udev (e.g. something similar to https://github.com/raspberrypi/picotool/pull/50 ) but that's not something that I'm familiar with myself.

aallan, 2024-01-30:
> > Any  ideas what's going on here, and how to fix it so that sudo is not required?
>
> I suspect that openocd needs root privileges to be able to talk to the hardware (i.e. the hardware that `interface/cmsisdap.cfg` is specifying)? That's probably fixable with udev (e.g. something similar to aspberrypi/picotool#50 ) but that's not something that I'm familiar with myself.

Yes, it'll need root access to access the hardware it's talking to, and also yes, it should be fixable with udev but I've never bothered looking into what would be necessary to do that.

peterharperuk, 2024-01-30:
Regarding running openocd. Are you in the plugdev group? Are you in the dialout group? I don't seem to need to run sudo and I don't appear to have a udev rule.

closed 2024-02-14 by peterharperuk

seamusdemora, 2024-02-14:
> I assume you've used the `pico_setup.sh` script? The `/home/pi/pico/openocd/tcl/target/rp2040.cfg` file is created by the "[git [...]

You assume correctly; i.e. "Yes, I used `pico_setup.sh` to "install" the SDK.

> > Any ideas what's going on here, and how to fix it so that `sudo` is not required?
> 
> I suspect that openocd needs root privileges to be able to talk to the hardware (i.e. the hardware that `interface/cmsis-dap.cfg` is specifying)? That's probably fixable with udev (e.g. something similar to [raspberrypi/picotool#50](https://github.com/raspberrypi/picotool/pull/50) ) but that's not something that I'm familiar with myself. 

Someone ("breaker") has formulated a solution for the `sudo` issue - it can be found in [this forum post](https://forums.raspberrypi.com/viewtopic.php?p=2193134#p2187755)...  (maybe you should hire this guy?). 

**My** solution was a [cop-out](https://dictionary.cambridge.org/dictionary/english/cop-out) - I simply created an *alias* for the full command - including `sudo` and `"adapter speed 5000"`.  

And BTW: 
I can understand why the `sudo` issue is complicated, but I fail to understand why the `"adapter speed 5000"` hasn't found an answer here (FWIW, I *tried* it in rpi2040.cfg without effect). I would think that among all of the domain-specific knowledge here, ***surely*** someone would know where to put this setting so that the SWD can upload the ELF file at a reasonable rate. One might get the impression that this is not the "A-team". So yes - by all means - let's close the issue.

peterharperuk, 2024-02-14:
Charming

seamusdemora, 2024-02-14:
> Charming

Hope you find the edited version a bit less offensive. But really Mr Harper - do you honestly feel this is the way things get fixed and improved?

peterharperuk, 2024-02-14:
Perhaps you can suggest to this b team member what fixes to this problem can be put the pico-sdk?

seamusdemora, 2024-02-15:
Perhaps you could actually **read** the comments? Like [this one](https://github.com/raspberrypi/pico-sdk/issues/1615#issuecomment-1916219797) from two weeks ago? The relevant part being: 

> And so then I tried the same edit in `/usr/local/share/openocd/scripts/target/rp2040.cfg`. It worked... this "did the trick". I feel like I've made progress, but wonder the point of putting two [...]

seamusdemora, 2024-02-15:
... and crickets.

lurch, 2024-02-15:
> but wonder the point of putting two copies of rp2040.cfg on the system

I already answered that:
The `/home/pi/pico/openocd/tcl/target/rp2040.cfg` file is created by the "[git clone](https://github.com/raspberrypi/pico-setup/blob/master/pico_setup.sh#L135)", and the `/usr/local/share/openocd/scripts/target/rp2040.cfg` file is created by the "[make install](https://github.com/raspberrypi/pico-setup/blob/master/pico_setup.sh#L140)".

> ... and crickets.

:cricket: :cricket_game:

seamusdemora, 2024-02-15:
@lurch 
Ah - so that answers the question re why there are two files, but does it also answer the question _"Why is the adapter speed config item is ignored in the file that's at `/home/pi/pico/openocd/tcl/target/rp2040.cfg`?"_

lurch, 2024-02-15:
Because when you run `openocd` at the command-line, that'll use the binary that exists in your `$PATH` (i.e. the one in `/usr/local/bin/` rather than the one in `/home/pi/pico/openocd/src/`) and as you might expect, the systemwide-installed version of `/usr/local/bin/openocd` will look for the corresponding systemwide-installed config files in `/usr/local/share/openocd/scripts/` .

I guess you also missed the part where I [said](https://github.com/raspberrypi/pico-sdk/issues/1615#issuecomment-1916471674):
If you're really concerned about there being "two copies of the file", you could delete the entire `/home/pi/pico/openocd` folder after the `pico_setup.sh` script has finished, if you want.

seamusdemora, 2024-02-15:
@lurch 

Even when the command line specifies: `sudo openocd -d0 ... -f target/rp2040.cfg` ...` ? What did it do with the `target` part?

lurch, 2024-02-15:
> What did it do with the `target` part?

**cleverca22** already answered that here: https://github.com/raspberrypi/pico-sdk/issues/1611#issuecomment-1923467090

> And I'm not really concerned about there being "two copies of the file" - I think you know that.

Sorry, perhaps I should have written "confused" rather than "concerned".

> The command I used comes from the "Getting Started" Guide:

If you have problems with the PDF documentation, the correct place for reporting that is at https://github.com/raspberrypi/pico-feedback not the `pico-sdk` repo.

seamusdemora, 2024-02-16:
> > What did it do with the `target` part?
> 
> **cleverca22** already answered that here: [#1611 (comment)](https://github.com/raspberrypi/pico-sdk/issues/1611#issuecomment-1923467090)
> 
> > And I'm not really concerned about there being "two copies of the file" - I think you know that.
> 
> Sorry, perhaps I should have written "confused" rather than "concerned".
> 
> > The command I used comes from the "Getting Started" Guide:
> 
> If you have problems with the PDF documentation, the correct place for reporting that is at https://github.com/raspberrypi/pico-feedback not the `pico-sdk` repo.

Oh, how very smart...  But how do we know the problem is with the documentation? It could be - as has been shown here - that the **location** of the problem is very hard to pin down!