raspberrypi/pico-sdk #1616: pio helper functions

peterharperuk, 2024-01-26:
It would be nice to add a pio_init function that picks a free pio, state machine and loads a program, as I keep writing this and it's easy to get wrong.

bool pio_init(const pio_program_t *program, PIO *pio_hw, uint *sm, uint *offset);
static void pio_deinit(const pio_program_t *program, PIO pio_hw, uint sm, uint offset);

and while we're at it, pio_get_instance like
static inline i2c_inst_t *i2c_get_instance(uint instance)

kilograham, 2024-01-26:
don't like the name `pio_init` since it doesnt initialize a PIO

peterharperuk, 2024-01-26:
pio_load?

lurch, 2024-01-27:
`pio_load_where_space_available` or `pio_load_if_possible` ? :wink: 

> and while we're at it, pio_get_instance like
> static inline i2c_inst_t *i2c_get_instance(uint instance)

Unfortunately that [function](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_i2c/include/hardware/i2c.h#L156) doesn't have any doxygen-comments, and thus doesn't appear in the [API documentation](https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#functions74).

peterharperuk, 2024-01-29:
some more ideas...

pio_claim_free_sm_add_program
pio_unclaim_sm_remove_program

peterharperuk, 2024-08-09:
These got added...
pio_claim_free_sm_and_add_program
pio_claim_free_sm_and_add_program_for_gpio_range
pio_remove_program_and_unclaim_sm
Some of the examples use them. At some point cyw43 will be updated to use them as well.

closed 2024-08-09 by peterharperuk