raspberrypi/pico-sdk #1617: Increasing stack size

niansa, 2024-01-27:
Hi! I have a project that operates entirely on stack (about 5 KB deep). However, the default stack size seems to be too low to comprehend this. How do I increase the stack size? Is this documented anywhere?

peterharperuk, 2024-01-27:
We could probably do with better documentation.

Assuming you're not using an OS like freertos then you'd have to tell us if you care about core 1. If we're talking about a program using core 0 only then I believe the answer is that you can increase stack to 4KB with PICO_STACK_SIZE. But practically you can keep increasing your stack usage until you collide with your heap. If that bothers you then you should take a look at writing your own linker script.

niansa, 2024-01-27:
Thanks! Since I don't use the heap (Idk if function like printf do though) it should be fine. Should I close this issue now?

Also, do you have documentation about integrating custom linker scripts, I can see this coming in handy for future projects of mine.

peterharperuk, 2024-01-29:
https://github.com/raspberrypi/pico-feedback/issues/372

closed 2024-01-29 by peterharperuk