raspberrypi/pico-sdk #1624: flash_safe_execute() not recognises FreeRTOS SMP define for the number of cores

antonwtw, 2024-02-03:
Hello, 
in pico-sdk/src/rp2_common/pico_flash/flash.c, on line 18,
when compiled with FreeRTOS SMP support,
`configNUM_CORES` is expected to be set to 2 through FreeRTOSConfig.h.
But the FreeRTOS uses a different name for this define: `configNUMBER_OF_CORES`.
See for instance line 100 in https://github.com/FreeRTOS/FreeRTOS-Community-Supported-Demos/blob/main/CORTEX_M0%2B_RP2040/Standard_smp/FreeRTOSConfig.h
This leads to an assert/error when calling flash_safe_execute() with FreeRTOS SMP, see line 172 in flash.c.
As a quick fix I had to add the following in my FreeRTOSConfig.h:
`#define configNUM_CORES configNUMBER_OF_CORES`
With best regards, Anton

lurch, 2024-02-03:
I guess this'll be fixed by #1530 ?

antonwtw, 2024-02-05:
That's true. Sorry I didn't find the 1530 before.

peterharperuk, 2024-02-14:
Duplicate

closed 2024-02-14 by peterharperuk