raspberrypi/pico-sdk #1627: SPI Init with defaul phase and polarization

juan0fran, 2024-02-06:
On spi.c [https://github.com/raspberrypi/pico-sdk/blob/2062372d203b372849d573f252cf7c6dc2800c0a/src/rp2_common/hardware_spi/spi.c#L21C6-L21C14]

there is not the option to change polarity/phase unless the library code is modified. 

A spi-specific function should be added (to have retro-compatibility for those libs already using this library) to something like:

uint spi_init_pol_phase(spi_inst_t *spi, uint baudrate, spi_cpol_t pol, spi_cpha_t phase) {
    spi_reset(spi);
    spi_unreset(spi);

    uint baud = spi_set_baudrate(spi, baudrate);
    spi_set_format(spi, 8, pol, phase, SPI_MSB_FIRST);
    // Always enable DREQ signals -- harmless if DMA is not listening
    hw_set_bits(&spi_get_hw(spi)->dmacr, SPI_SSPDMACR_TXDMAE_BITS | SPI_SSPDMACR_RXDMAE_BITS);

    // Finally enable the SPI
    hw_set_bits(&spi_get_hw(spi)->cr1, SPI_SSPCR1_SSE_BITS);
    return baud;
}

lurch, 2024-02-06:
> there is not the option to change polarity/phase unless the library code is modified.

Can't you simply call `spi_set_format` after calling the existing `spi_init` ?

juan0fran, 2024-02-06:
Actually yes, but that would require to define `spi_set_format` as non-static. Either way would work.

peterharperuk, 2024-02-14:
spi_set_format is inline, you should be able to call it

kilograham, 2024-05-18:
closing as you can indeed call `spi_set_format`; please re-open if this causes other issues

closed 2024-05-18 by kilograham