raspberrypi/pico-sdk #1631: ability to set git tag to download in external/pico_sdk_import.cmake

ahmetkizilay, 2024-02-08:
it would be useful to lock the release/tag for projects that depends on the pico-sdk. 

We can use PICO_SDK_FETCH_GIT_TAG to specify a release tag for the Pico SDK to download.

example usage:
```
set(PICO_SDK_FETCH_FROM_GIT true)
set(PICO_SDK_FETCH_FROM_GIT_TAG "1.5.1")
include(pico_sdk_import.cmake)
```

peterharperuk, 2024-02-08:
I don't think we'd use it. master is always the latest release

lurch, 2024-02-08:
I guess the implied usage-model here is for people who are copying `pico_sdk_import.cmake` into their own project, and want to pin to specific version of pico-sdk, so that their project doesn't get accidentally broken by later pico-sdk updates?

peterharperuk, 2024-02-08:
That would make some sense I guess. Although isn't that what git submodules are for?

kilograham, 2024-06-05:
merged into devleop

PR #1630 (2024-06-05): set git tag for the SDK
with this change we can use PICO_SDK_FETCH_GIT_TAG to specify a release tag for the Pico SDK to download.

example usage:
```
set(PICO_SDK_FETCH_FROM_GIT true)
set(PICO_SDK_FETCH_FROM_GIT_TAG "1.5.1")
include(pico_sdk_import.cmake)
```

Fixes #1631
closed 2024-06-05 by kilograham