raspberrypi/pico-sdk #1632: TinyUSB Toggle between HID_PROTOCOL_BOOT and HID_PROTOCOL_REPORT

moc32, 2024-02-08:
Toggling mouse interface protocol between boot and report on Logitech K270/M185 blocks the ongoing interrupt endpoint transfer. This works for several KM devices. The only difference between those device and the Logitech K270/M185 or K220/M150 is the the wMaxPacketSize parameter. On the Logitech combo devices the value is 20 bytes for mouse endpoint and 8 bytes for keyboard endpoint. For the other devices the wMaxPacketSize is less than or equal to 8 bytes.

I have modified the tinyusb bare api example to add the protocol toggling.

`/*
 * The MIT License (MIT)
 *
 * Copyright (c) 2019 Ha Thach (tinyusb.org)
 *
 * Permission is hereby granted, free of charge, to any person obtaining a copy
 * of this software and associated documentation files (the "Software"), to deal
 * in the Software without restriction, including without limitation the rights
 * to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
 * copies of the Software, and to permit persons to whom the Software is
 * furnished to do so, subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in
 * all copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 * FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
 * AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 * LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
 * OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
 * THE SOFTWARE.
 *
 */

/* This example current worked and tested with following controller
 * - Sony DualShock 4 [CUH-ZCT2x] VID = 0x054c, PID = 0x09cc
 */

#include <stdlib.h>
#include <stdio.h>
#include <string.h>

#include "bsp/board.h"
#include "tusb.h"

// English
#define LANGUAGE_ID 0x0409
#define BUF_COUNT 4
#define DEV_COUNT 3
#define ENDPOINT_COUNT 3
#define INTERFACE_COUNT 3

tusb_desc_device_t desc_device;

uint8_t buf_pool[BUF_COUNT][64];
uint8_t buf_owner[BUF_COUNT] = {0}; // device address that owns buffer

uint8_t wMaxPacketSize[DEV_COUNT][ENDPOINT_COUNT] = {{0, 0, 0}, {0, 0, 0}, {0, 0, 0}};
uint8_t bEndpointAddress[DEV_COUNT][ENDPOINT_COUNT] = {{0, 0, 0}, {0, 0, 0}, {0, 0, 0}};
tuh_xfer_t xferEndpoint[DEV_COUNT][ENDPOINT_COUNT];
uint8_t bInterfaceProtocol[DEV_COUNT][INTERFACE_COUNT] = {{0, 0, 0}, {0, 0, 0}, {0, 0, 0}};

//--------------------------------------------------------------------+
// MACRO CONSTANT TYPEDEF PROTYPES
//--------------------------------------------------------------------+
void led_blinking_task(void);

static void print_utf16(uint16_t *temp_buf, size_t buf_len);
void print_device_descriptor(tuh_xfer_t *xfer);
void parse_config_descriptor(uint8_t dev_addr, tusb_desc_configuration_t const *desc_cfg);

uint8_t *get_hid_buf(uint8_t daddr);
void free_hid_buf(uint8_t daddr);

//
[... 534 lines cut ...]
                  .direction = TUSB_DIR_OUT},
          .bRequest = HID_REQ_CONTROL_SET_PROTOCOL,
          .wValue = protocol,
          .wIndex = interface - 1,
          .wLength = 0};

  tuh_xfer_t xfer =
      {
          .daddr = daddr,
          .ep_addr = 0,
          .setup = &request,
          .buffer = NULL,
          .complete_cb = HIDSetRequestDone,
          .user_data = (uintptr_t)0};

  printf("HID Set: Interface=%d, Protocol = %d\r\n", interface, protocol);

  if (!tuh_control_xfer(&xfer))
  {
    printf("Failed to set protocol\r\n");
    return false;
  }

  return true;
}`

moc32, 2024-02-08:
This issue might look trivial but It looks like an issue with the hardware controller. After toggling the report protocol, the keyboard endpoint get blocked. The host sends request to the keyboard endpoint and the keyboard is replying the IN request but the host is not sending the ACK and restart the IN transfer without firing any interrupt.

peterharperuk, 2024-02-08:
@P33M does this make any sense to you?

moc32, 2024-02-08:
<
//--------------------------------------------------------------------+
// Device Descriptor
//--------------------------------------------------------------------+

void print_device_descriptor(tuh_xfer_t *xfer)
{
  if (XFER_RESULT_SUCCESS != xfer->result)
  {
    printf("Failed to get device descriptor\r\n");
    return;
  }

  uint8_t const daddr = xfer->daddr;

  printf("Device %u: ID %04x:%04x\r\n", daddr, desc_device.idVendor, desc_device.idProduct);
  printf("Device Descriptor:\r\n");
  printf("  bLength             %u\r\n", desc_device.bLength);
  printf("  bDescriptorType     %u\r\n", desc_device.bDescriptorType);
  printf("  bcdUSB              %04x\r\n", desc_device.bcdUSB);
  printf("  bDeviceClass        %u\r\n", desc_device.bDeviceClass);
  printf("  bDeviceSubClass     %u\r\n", desc_device.bDeviceSubClass);
  printf("  bDeviceProtocol     %u\r\n", desc_device.bDeviceProtocol);
  printf("  bMaxPacketSize0     %u\r\n", desc_device.bMaxPacketSize0);
  printf("  idVendor            0x%04x\r\n", desc_device.idVendor);
  printf("  idProduct           0x%04x\r\n", desc_device.idProduct);
  printf("  bcdDevice           %04x\r\n", desc_device.bcdDevice);

  // Get String descriptor using Sync API
  uint16_t temp_buf[256];

  printf("  iManufacturer       %u     ", desc_device.iManufacturer);
  if (XFER_RESULT_SUCCESS == tuh_descriptor_get_manufacturer_string_sync(daddr, LANGUAGE_ID, temp_buf, sizeof(temp_buf)))
  {
    print_utf16(temp_buf, TU_ARRAY_SIZE(temp_buf));
  }
  printf("\r\n");

  printf("  iProduct            %u     ", desc_device.iProduct);
  if (XFER_RESULT_SUCCESS == tuh_descriptor_get_product_string_sync(daddr, LANGUAGE_ID, temp_buf, sizeof(temp_buf)))
  {
    print_utf16(temp_buf, TU_ARRAY_SIZE(temp_buf));
  }
  printf("\r\n");

  printf("  iSerialNumber       %u     ", desc_device.iSerialNumber);
  if (XFER_RESULT_SUCCESS == tuh_descriptor_get_serial_string_sync(daddr, LANGUAGE_ID, temp_buf, sizeof(temp_buf)))
  {
    print_utf16(temp_buf, TU_ARRAY_SIZE(temp_buf));
  }
  printf("\r\n");

  printf("  bNumConfigurations  %u\r\n", desc_device.bNumConfigurations);

  // Get configuration descriptor with sync API
  if (XFER_RESULT_SUCCESS == tuh_descriptor_get_configuration_sync(daddr, 0, temp_buf, sizeof(temp_buf)))
  {
    parse_config_descriptor(daddr, (tusb_desc_configuration_t *)temp_buf);
  }
}
/>

moc32, 2024-02-13:
Hello @peterharperuk, @P33M @hathach,
Do you have any idea of what is going on here? 
You can easily reproduce the issue by building the attached example project.
After building the project, you can load the UF2 file on the pico board then connect a Logitech wireless KM (ex. K270/M185 or K220/M150) and click "R" to switch mouse to report mode or "B" to switch it to boot mode.
You can also connect a wired keyboard using a hub since the PICO has only one USB port and in this case the switching works. The switching works also with combo devices from microsoft.
The only difference between the logitech combo devices and other tested devices is that the the mouse interface wMaxPacketSize is 20 bytes where it is 8bytes or less for other devices.

kilograham, 2024-05-19:
moving to none until we know this is an SDK issue