raspberrypi/pico-sdk #1638: allowing label to be used in move instructions

webdeb, 2024-02-17:
Hello,

I am working on a software which uses multiple `StateMachines` and want to share some generic instructions for many state machines instances.

I would like to jump to an instruction in memory which is globaly known, but then I have to jmp back into the instruction of the current program. The problem is, that its impossible to know to which instruction the program should jump to.

So the idea is allow to cp the compiled label() result into a scratch register value, so it will jmp there.

```
.label "common_load"
  pull()
  move(isr, osr)
  pull()
  jmp(x)
  

  # some other sm
  move(x, label("after_load"))
  jmp("common_load")
  label("after_load")
```

lurch, 2024-02-18:
From having a quick flick through Chapter 3 of https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf you _might_ be able to do your `move(x, label("after_load"))` thing with labels and the SET instruction? :shrug:

webdeb, 2024-02-18:
Thx @lurch, I hope so. there is also more about absolut jumps in the directive sections 3.3.1.
I was wondering, if there are some example for this, or advises, because 32 instructions limit and common programs are challenging topics, _I guess_.

lurch, 2024-02-18:
The pico-sdk issues section on GitHub is just for reporting bugs. For support requests / general discussion, you should use the forums https://forums.raspberrypi.com/

There's some PIO example code in https://github.com/raspberrypi/pico-examples/tree/master?tab=readme-ov-file#pio but I've no idea if any of those do the specific thing you're asking for.

closed 2024-02-18 by webdeb