raspberrypi/pico-sdk #1639: Wrong value for PICO_SDK_PATH after automatic downloadi from git

stefanschober, 2024-02-19:
I'm configuring pico_sdk from my project with the following settings:
- PICO_SDK_FETCH_FROM_GIT set to TRUE
- PICO_SDK_FETCH_FROM_GIT_PATH set to the absolute target path for the pico_sdk

When configuring the project, pico_sdk is downloaded into ${PICO_SDK_FETCH_FROM_GIT_PATH}/pico_sdk-src. The value of PICO_SDK_PATH is however set to ${PICO_SDK_FETCH_FROM_GIT_PATH}.

I fixed this like so:

```
--- ../pico-sdk/pico_sdk-src/external/pico_sdk_import.cmake     2024-02-19 10:23:26.764972800 +0100
+++ cmake/pico_sdk_import.cmake 2024-02-19 11:47:44.588034900 +0100
@@ -58,14 +59,20 @@
     endif ()
 endif ()

-get_filename_component(PICO_SDK_PATH "${PICO_SDK_PATH}" REALPATH BASE_DIR "${CMAKE_BINARY_DIR}")
+file(REAL_PATH "${PICO_SDK_PATH}" PICO_SDK_PATH BASE_DIRECTORY "${CMAKE_BINARY_DIR}")
 if (NOT EXISTS ${PICO_SDK_PATH})
     message(FATAL_ERROR "Directory '${PICO_SDK_PATH}' not found")
 endif ()

-set(PICO_SDK_INIT_CMAKE_FILE ${PICO_SDK_PATH}/pico_sdk_init.cmake)
+find_file(PICO_SDK_INIT_CMAKE_FILE pico_sdk_init.cmake
+    HINTS ${PICO_SDK_PATH}
+    PATH_SUFFIXES pico_sdk-src
+)
+# set(PICO_SDK_INIT_CMAKE_FILE ${PICO_SDK_PATH}/pico_sdk_init.cmake)
 if (NOT EXISTS ${PICO_SDK_INIT_CMAKE_FILE})
     message(FATAL_ERROR "Directory '${PICO_SDK_PATH}' does not appear to contain the Raspberry Pi Pico SDK")
+else()
+    cmake_path(GET PICO_SDK_INIT_CMAKE_FILE PARENT_PATH PICO_SDK_PATH)
 endif ()

 set(PICO_SDK_PATH ${PICO_SDK_PATH} CACHE PATH "Path to the Raspberry Pi Pico SDK" FORCE)
```

stefanschober, 2024-02-21:
happens when re-configuring the project using `cmake ... --fresh`

will-v-pi, 2024-07-23:
@stefanschober I can't reproduce this, but it could be a cmake caching issue. Does replacing `set(PICO_SDK_PATH ${pico_sdk_SOURCE_DIR})` with `set(PICO_SDK_PATH ${pico_sdk_SOURCE_DIR} CACHE PATH "Path to the Raspberry Pi Pico SDK" FORCE)` (line 51 of that file) fix the issue for you?

The `pico_sdk_SOURCE_DIR` variable is set correctly to `${PICO_SDK_FETCH_FROM_GIT_PATH}/pico_sdk-src` by FetchContent, so it should set the `PICO_SDK_PATH` to the correct value

kilograham, 2025-07-21:
closing as old and not reproduced