raspberrypi/pico-sdk #1641: IS25LP***D ID size is wrong in SDK and produce duplicated flash uID

eNovware, 2024-02-19:
Hi world,

I'm working with custom RP2040 boards with IS25LP***D flash chip (usually IS25LP016D)

I've made custom board file and all works fine.

The trouble is with the uID returned using **_pico_get_unique_board_id( )_** and **_pico_get_unique_board_id_string( )_** functions.

**_pico_unique_board_id_t_** is 8 bytes long but the uID from IS25LP***D is 16 bytes long, so we have a lot of same uID.

Maybe could I override PICO_UNIQUE_BOARD_ID_SIZE_BYTES in my board file ?
... any idea to solve that ?

Thansk for reading.
Sebastien.

peterharperuk, 2024-02-19:
@kilograham Is there a reason why PICO_UNIQUE_BOARD_ID_SIZE_BYTES is not overridable or was it an oversight?

eNovware, 2024-02-19:
Looks like this ID size is near hard-coded in **_src/rp2_common/hardware_flash/flash.c_**

This should be linked to flash selected instead.

Maybe it's easier in my case to try to read ID with custom function but since flash store my code
... probably not as simple as to write it ...

eNovware, 2024-02-20:
What about something like this

```
void flash_get_unique_id_IS25LP(uint8_t *id_out)
{
  const int IS25LP_FLASH_RUID_CMD 0x4b
  const int IS25LP_FLASH_RUID_DUMMY_BYTES 4
  const int IS25LP_FLASH_RUID_DATA_BYTES 16
  const int IS25LP_FLASH_RUID_TOTAL_BYTES (1 + IS25LP_FLASH_RUID_DUMMY_BYTES + IS25LP_FLASH_RUID_DATA_BYTES)
  //
  uint8_t txbuf[IS25LP_FLASH_RUID_TOTAL_BYTES] = {0};
  uint8_t rxbuf[IS25LP_FLASH_RUID_TOTAL_BYTES] = {0};
  //
  txbuf[0] = IS25LP_FLASH_RUID_CMD;
  flash_do_cmd(txbuf, rxbuf, IS25LP_FLASH_RUID_TOTAL_BYTES);
  //
  for (int i = 0; i < IS25LP_FLASH_RUID_DATA_BYTES; i++)
    id_out[i] = rxbuf[i + 1 + IS25LP_FLASH_RUID_DUMMY_BYTES];
}
```

But looks like calls to that kind of low-level flash RW must to be done as soon à possible.

The fisrt line in my main() is `stdio_init_all()`, should I call flash_get_unique_id_IS25LP() before or just after it ?
( or does `stdio_init_all()` in `main()` is already too late in CPU initialization ?!! )

Thanks.

lurch, 2024-02-20:
Looks like this code was initially added in https://github.com/raspberrypi/pico-sdk/commit/c1196e9af61edd374db42412a8b720d15afc152f which says "serial NOR flash devices which have a 64-bit unique ID as a standard feature" ?

eNovware, 2024-02-20:
That's the point : some flash have 128b uID and other 64b, but SDK actually only read 64b uID, so all (a lot of !) 128b flashs chip have same uID.

lurch, 2024-02-20:
...which I assume means that when the code was first written, we were only testing with flash chips with a 64b uID :wink:

eNovware, 2024-02-20:
I fully understand why it's done as it's done, but how could I get my full uID ?

Does the function in upper post have any chance to work or do I have to wait for SDK to fix this ?

The SDK says : << [...] executing flash code on the second core concurrently will be fatal. To avoid these pitfalls it is recommended that this function only be used to extract flash metadata during startup, before the main application begins to run: see the implementation of pico_get_unique_id() for an example of this. >>

Pretty terrifying !!!

I can run this once in my main() before or after the **_stdio_init_all()_**, but is it safe ?
( really safe I mean, not just "it will works most of the time" :) )

kilograham, 2024-02-20:
you can call it (without additional protection) if you aren't running anything on core 1 (which you won't be unless/until you started something yourself)

peterharperuk, 2024-02-20:
A quick look at the datasheet suggests that might work. You should be ok calling it in main - nothing runs on core1 by default. Yes, you'd need a change to the sdk for a proper fix.

lurch, 2024-08-13:
#1132 points out that PY25Q32 also has a 128-bit (rather than 64-bit) unique-ID.

lurch, 2024-08-15:
You might like to test #1819 ?

PR #1819 (2024-08-15): Support serial flash with 16 byte unique id
Flash devices like IS25LP016D have a 16 byte unique id. Reading just the first 8 bytes causes devices to have duplicate unique ids as the numbers only differ in the final 8 bytes.

Support overriding FLASH_UNIQUE_ID_SIZE_BYTES in the board header. Use the final 8 bytes for the unique id.

Fixes #1641
commit e049f506 (2024-08-15): Support serial flash with 16 byte unique id (#1819)

* Support serial flash with 16 byte unique id

Flash devices like IS25LP016D have a 16 byte unique id. Reading just the
first 8 bytes causes devices to have duplicate unique ids as the numbers
only differ in the final 8 bytes.

Support overriding FLASH_UNIQUE_ID_SIZE_BYTES in the board header. Use
the final 8 bytes for the unique id.

Fixes #1641
closed 2024-08-15 by peterharperuk