raspberrypi/pico-sdk #1646: Using FreeRTOS-Kernel-Heap3 library leads to deadlock across cores, race condition on core spinlock

gemarcano, 2024-02-23:
I describe the full issue here: https://forums.freertos.org/t/multicore-smp-rp2040-port-deadlock-when-using-heap-3/19255/

In summary, FreeRTOS heap_3 calls `vTaskSuspendAll` which grabs the task lock before calling malloc/free. At the same time, if another task is using good-ol' malloc/free, a race takes place. Under FreeRTOS, the rp2040 port uses some macro magic to inject FreeRTOS support into the mutex functions. This implementation tries to grab the task lock also. heap_3 takes the FreeRTOS task lock before grabbing `malloc_mutex`, while a normal call to `malloc` takes the core synchronization spinlock first before grabbing the FreeRTOS task lock. This leads to a deadlock between the two tasks.

For clarity, let's say `core 0` is calling into FreeRTOS heap_3, and `core 1` is calling `malloc` (which is wrapped by pico-sdk). This is the sequence of events:

1. `core 0`: `vTaskSuspendAll` in `vPortFree` before calling `free`
2. `core 1`: calls `malloc`
3. `core 1`: in `mutex_enter_blocking` with `malloc_mutex`, grabs core synchronization spinlock
4. `core 0` calls `free`
5. `core 0`: in `mutex_exit` with `malloc_mutex`, blocks waiting for core synchronization spinlock
6. `core 1`: calls `lock_internal_spin_unlock_with_wait` which is a macro provided by FreeRTOS rp2040 port
7. `core 1`: tries to grab task lock with an eventual call to `vTaskEnterCritical` after macro hell
8. deadlock, `core 1` is holding onto the core synchronization spinlock, `core 0` has the task lock.

This happens because FreeRTOS heap_3 is trying to be multi task safe, but the pico-sdk's malloc and free wrappers are already providing this. I don't know of a good way to fix/clean this up, as it will likely need better cooperation between the rp2040 port and FreeRTOS itself.

My current workaround is to waste more SRAM and just use heap_4 and let FreeRTOS manage its own heap for its own objects, and let tasks use the "regular" heap that uses pico-sdk locking.

gemarcano, 2024-02-23:
I didn't make it clear, but I am seeing this on an actual project, so it's not just a hypothetical.

peterharperuk, 2024-02-23:
Can you post what version of freertos you're using?
And any chance of an example that demonstrates the problem?

kilograham, 2024-02-24:
possibly the same issue as #1453

NewmanIsTheStar, 2024-02-25:
Could this bug cause also cause the following assertion?
**assertion "pxYieldSpinLock[get_core_num()] == ((void *)0)" failed: file "port.c", line 363**

kilograham, 2024-08-09:
this should be fixed with the latest SDK and FreeRTOS-Kernel (https://github.com/raspberrypi/FreeRTOS-Kernel

closed 2024-08-09 by kilograham