raspberrypi/pico-sdk #1647: Get the SDK and exemples: Failure to download and initialise submodule

samdestroy3, 2024-02-25:
In section 2.1: Get the SDK and examples.

$ git clone https://github.com/raspberrypi/pico-sdk.git --branch master
$ cd pico-sdk
$ git submodule update --init

The command submodule update --init does not work.
[image]

lurch, 2024-02-25:
I just tested and it works fine here - maybe you just had a temporary network glitch? :shrug: 

[image]

samdestroy3, 2024-02-25:
Thank's for the validation. It may be as you said a network issues. I'm still having the issue. I Will try on an other network

closed 2024-02-25 by samdestroy3