raspberrypi/pico-sdk #1652: Wrong compiler selected (cc instead of arm-none-eabi-gcc)

arminveres, 2024-02-26:
I have been having this issue on and off, where `/usr/bin/cc` is selected, instead of `/opt/gcc-arm-none-eabi/bin` which I even provide through `PICO_TOOLCHAIN_PATH`.
I honestly do not understand why this could be a problem. The toolchain is on the path in any case.
Further, I also tried setting `CMAKE_CXX_COMPILER` but then I get the error, seen in #1597 .

Isn't there a cleaner or simpler way to make it work?
[image]

lurch, 2024-02-26:
What platform / OS / version are you building on? Are you following the instructions in https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf ?

arminveres, 2024-02-29:
I am running Arm GNU toolchain directly downloaded from [Arm](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads). OS is Fedora WS 39 and I have tried running on `master` and `1.5.1`.

I found out I wasn't including it always before the Project, as I am running multiple libraries that are linking against it, so it was an oversight on my part.

closed 2024-02-29 by arminveres