raspberrypi/pico-sdk #1658: set_sys_clock_khz not work under 20 MHz with sdk 1.5.1

hodlemil, 2024-03-04:
set_sys_clock_khz not work under 20000 kHz. I try with 19000,12000,10000 with sdk version 1.5.1. The attached source code work without problem with sdk version 1.3.1, but not work with sdk version 1.5.1. What changes between sdk version 1.3.1 and 1.5.1. to make int notwork?

[set_sys_clock_rp2040.zip](https://github.com/raspberrypi/pico-sdk/files/14484759/set_sys_clock_rp2040.zip)

peterharperuk, 2024-03-05:
Any idea @andygpz11

hodlemil, 2024-03-05:
The answer was in the changelog.
In cmakelists.txt
target_compile_definitions(cam-sw PRIVATE
PICO_PLL_VCO_MIN_FREQ_MHZ=400
)
solved the problem, because new sdk set this define to 700 the 1.3.1 sdk sets to 400

closed 2024-03-05 by hodlemil