raspberrypi/pico-sdk #1659: FreeRTOS v11 support

eamars, 2024-03-05:
Hello All
Given the FreeRTOS v11 is released at the end of 2023, the pico-sdk,especially the functions that supports SMP, need to be updated to support the new flag to enable [SMP](https://freertos.org/symmetric-multiprocessing-introduction.html). 

For example 
https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_async_context/async_context_freertos.c#L15

The code 

    #if configNUM_CORES > 1 && !defined(configUSE_CORE_AFFINITY)
    #error async_context_freertos requires configUSE_CORE_AFFINITY under SMP
    #endif

Should be updated to look for the new config `configNUMBER_OF_CORES`. 
There are more files need to be updated. 

Reference: 
https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/smp/README.md

peterharperuk, 2024-03-05:
https://github.com/raspberrypi/pico-sdk/pull/1530

We suspect there might be more work to do

eamars, 2024-03-05:
Thanks for pointing this out @peterharperuk 
I'm testing the compatibility of FreeRTOS v11 with Picosdk right now and haven't found any issues so far.

closed 2024-08-08 by kilograham

behines, 2024-09-01:
I've been fighting for two days with FreeRTOS V11.1.0 and Pico-SDK 2.0.0, aided by the Pico VS Code Extension.

When trying to compile VS Code, I would get "cannot find file hardware/sync.h" while trying to compile various modules, like event_groups.c.

I am new to CMake, sospent two days learning the basics, and was finally able to get past this and similar errors by adding this to my top-level CMakeLists.txt:

```
target_include_directories(freertos_kernel PUBLIC SYSTEM
  ${PICO_SDK_PATH}/src/rp2_common/hardware_sync/include
  ${PICO_SDK_PATH}/src/rp2_common/hardware_sync_spin_lock/include
  ${PICO_SDK_PATH}/src/rp2_common/pico_runtime_init/include
)
```

I don't understand why this isn't handled by the FreeRTOS RP2040 extension?  It seems like maybe it's not up to date with the rest of FreeRTOS, but then I found this bug report #1659 that seems to indicate that this should work?

In any event, adding those lines to my CMakeLists.txt only got me partway to the finish line. I am still striving mightily but am now getting this when compiling the FreeRTOS RP2040 port's port.c:

```
In file included from C:/Users/brad/Documents/PlanetA/Software/PicoStarting/blinkRTOS/build/_deps/freertos_kernel-src/portable/ThirdParty/GCC/RP2040/port.c:129:
[build] C:/Users/brad/Documents/PlanetA/Software/PicoStarting/blinkRTOS/build/_deps/freertos_kernel-src/portable/ThirdParty/GCC/RP2040/port.c: In function 'vPortLockInternalSpinUnlockWithNotify':
[build] C:/Users/brad/Documents/PlanetA/Software/PicoStarting/blinkRTOS/build/_deps/freertos_kernel-src/include/event_groups.h:619:5: warning: implicit declaration of function 'xTimerPendFunctionCallFromISR' [-Wimplicit-function-declaration]
[build]   619 |     xTimerPendFunctionCallFromISR( vEventGroupSetBitsCallback, ( void * ) ( xEventGroup ), ( uint32_t ) ( uxBitsToSet ), ( pxHigherPriorityTaskWoken ) )
[build]       |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[build] C:/Users/brad/Documents/PlanetA/Software/PicoStarting/blinkRTOS/build/_deps/freertos_kernel-src/portable/ThirdParty/GCC/RP2040/port.c:1067:17: note: in expansion of macro 'xEventGroupSetBitsFromISR'
[build]  1067 |                 xEventGroupSetBitsFromISR( xEventGroup, uxBits, &xHigherPriorityTaskWoken );
[build]       |                 ^~~~~~~~~~~~~~~~~~~~~~~~~
[build] C:/Users/brad/Documents/PlanetA/Software/PicoStarting/blinkRTOS/build/_deps/freertos_kernel-src/portable/ThirdParty/GCC/RP2040/port.c:1091:17: error: 'sio_hw' undeclared (first use in this function); did you mean 'scb_hw'?
[build]  1091 |                 sio_hw->fifo_wr = 0;
[build]       |                 ^~~~~~
[build]       |                 scb_hw
```

I started to feel like maybe this was just versionitis, so I attempted to do the build with pico-sdk 1.4.0, since it slightly predates the last major FreeRTOS commit, but then I got toolchain failures.  Probably fixable but seemed like a deep hole.  So I'm back to trying to push through with 2.0.0.

Am I missing something?  Is the SDK 2.0.0 believed to work with FreeRTOS 11.1?

Thanks,
Brad

kilograham, 2024-09-01:
> target_include_directories(freertos_kernel PUBLIC SYSTEM
  ${PICO_SDK_PATH}/src/rp2_common/hardware_sync/include
  ${PICO_SDK_PATH}/src/rp2_common/hardware_sync_spin_lock/include
  ${PICO_SDK_PATH}/src/rp2_common/pico_runtime_init/include
)

You should not have to do any of that.

First step; Make sure you delete your build directories when changing SDK or FreeRTOS versions.

I would use FreeRTOS from here: https://github.com/raspberrypi/FreeRTOS-Kernel for best results

behines, 2024-09-01:
Hi Kilograham,

Thanks for the super-fast response!  Much much appreciated!

Thanks for the tip on building, I have indeed been deleting the entire build folder every few minutes as I work on this.  

However, I have been using a different distro of FreeRTOS. I've been getting it from here: https://github.com/FreeRTOS/FreeRTOS-Kernel.git

So I completely blew away the build folder and used your recommended distro instead, then did CMake Configure then Build, but I get the exact same error, first appearing in the compile of croutine.c:

```
[build] In file included from C:/Users/brad/Documents/PlanetA/Software/PicoStarting/blinkRTOS/build/_deps/freertos_kernel-src/include/portable.h:53,
[build]                  from C:/Users/brad/Documents/PlanetA/Software/PicoStarting/blinkRTOS/build/_deps/freertos_kernel-src/include/FreeRTOS.h:101,
[build]                  from C:/Users/brad/Documents/PlanetA/Software/PicoStarting/blinkRTOS/build/_deps/freertos_kernel-src/croutine.c:29:
[build] C:/Users/brad/Documents/PlanetA/Software/PicoStarting/blinkRTOS/build/_deps/freertos_kernel-src/portable/ThirdParty/GCC/RP2040/include/portmacro.h:40:10: fatal error: hardware/sync.h: No such file or directory
[build]    40 | #include "hardware/sync.h"
[build]       |          ^~~~~~~~~~~~~~~~~
```

I wonder if my CMakeLists.txt is correct.  Below is what I am using.  I couldn't find any all-in-one example of a CMakeLists.txt to use, so I pieced this together by starting from the skeleton provided by the VS Code Pico extension, then augmenting with the instructions I found at the FreeRTOS link you provided and the [RP2040 port page](https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/main/portable/ThirdParty/GCC/RP2040).  (Note that the error is the same regardless of whether I specify board type "PICO" or "adafruit_itsybitsy_rp2040".)

Thanks so much for any advice you can offer,
Brad

```
cmake_minimum_required(VERSION 3.13)

# Set project data
set(PROJECT_NAME "RP2040_TEST_PROJECT")
set(BUILD_NUMBER "1")

# We put our FreeRTOSConfig.h in here
set(FREERTOS_CFG_DIRECTORY "FreeRTOS_config")

set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

# == DO NEVER EDIT THE NEXT LINES for Raspberry Pi Pico VS Code Extension to work ==
if(WIN32)
    set(USERHOME $ENV{USERPROFILE})
else()
    set(USERHOME $ENV{HOME})
endif()
set(sdkVersion 2.0.0)
set(toolchainVersion 13_2_Rel1)
set(picotoolVersion 2.0.0)
include(${USERHOME}/.pico-sdk/cmake/pico-vscode.cmake)
# ====================================================================================

set(PICO_BOARD adafruit_itsybitsy_rp2040 CACHE STRING "Board type")
#set(PICO_BOARD PICO CACHE STRING "Board type")

include(pico_sdk_import.cmake)

# Name the project
project(${PROJECT_NAME}
  LANGUAGES C CXX ASM
  DESCRIPTION "FreeRTOS-based RP2040 test"
)

# Initialise the Pico SDK
pico_sdk_init()

#####
# FreeRTOS
#

add_library(freertos_config INTERFACE)
target_compile_definitions(freertos_config
  INTERFACE
    projCOVERAGE_TEST=0
    ${definitions}
)
target_compile_options(freertos_config INTERFACE ${options})
target_include_directories(freertos_config SYSTEM
  INTERFACE
    ${FREERTOS_CFG_DIRECTORY}
)

# Select Heap 4 and the RP2040 port
set(FREERTOS_HEAP "4" CACHE STRING "FreeRTOS Heap Type" FORCE)
set(FREERTOS_PORT "GCC_RP2040" CACHE STRING "FreeRTOS port folder" FORCE)

# Pull in FreeRTOS
include(FetchContent)  

FetchContent_Declare(
  freertos_kernel
  GIT_REPOSITORY https://github.com/raspberrypi/FreeRTOS-Kernel.git
  GIT_TAG        4f7299d  # Freeze on this specific commit from August 2024
)

FetchContent_MakeAvailable(freertos_kernel)  

# Set the FREERTOS_KERNEL_PATH to the fetched content directory
# The RP2040 CMakeLists.txt uses this as well as us using it
set (FREERTOS_KERNEL_PATH ${freertos_kernel_SOURCE_DIR} CACHE STRING "FreeRTOS Kernel Path" FORCE)
set (RP2040_PORT_PATH ${FREERTOS_KERNEL_PATH}/portable/ThirdParty/GCC/RP2040 CACHE STRING "FreeRTOS RP2040 Port Path" FORCE)

add_subdirectory(${RP2040_PORT_PATH})

###########
# The executable
#

add_executable(${PROJECT_NAME}
  blink.c
)

# pull in common dependencies
target_link_libraries(${PROJECT_NAME}  
  freertos_kernel
  pico_stdlib 
)

# Enable/disable STDIO via USB and UART
pico_enable_stdio_usb(${PROJECT_NAME} 1)
pico_enable_stdio_uart(${PROJECT_NAME} 1)

# create map/bin/hex file etc.
pico_add_extra_outputs(${PROJECT_NAME})

```

behines, 2024-09-01:
A little follow-up on what I've learned.  I've limped my way to a working system.  Not pretty but it's working.

Adding these lines will give me a successful build:

```
target_include_directories(freertos_kernel PUBLIC SYSTEM
  ${PICO_SDK_PATH}/src/rp2_common/hardware_sync/include
  ${PICO_SDK_PATH}/src/rp2_common/hardware_sync_spin_lock/include
)

# The above only works for files that are part of freertos_kernel.  Other files like 
# irq.c are pulled in from the Pico SDK and use a different nclude list.  This burnishes
# the list for each of those so that they can compile.
target_include_directories(hardware_irq SYSTEM INTERFACE
  ${PICO_SDK_PATH}/src/rp2_common/pico_runtime/include
  ${PICO_SDK_PATH}/src/rp2_common/pico_runtime_init/include
)
target_include_directories(hardware_watchdog SYSTEM INTERFACE
  ${PICO_SDK_PATH}/src/rp2_common/hardware_boot_lock/include
 ${PICO_SDK_PATH}/src/rp2_common/pico_bootrom/include
)
```

*if* I also add these lines to FreeRTOSConfig.h:

```
/* Use Pico SDK ISR handlers */
#define vPortSVCHandler isr_svcall
#define xPortPendSVHandler isr_pendsv
#define xPortSysTickHandler isr_systick

/* Pico customizations needed to get it to compile.  See rp2040_config.h */
#define configUSE_DYNAMIC_EXCEPTION_HANDLERS  0
#define configSUPPORT_PICO_SYNC_INTEROP       0
#define configSUPPORT_PICO_TIME_INTEROP       0

#define configTOTAL_HEAP_SIZE                 4096
```

I have managed to get a single-task Blinky test program running with this build.

But of course I'd much prefer to follow a cleaner, approved approach.

Brad

kilograham, 2024-09-02:
So it seems the majority of the issue is with the way your are importing FreeRTOS-Kernel. You are using their top level CMakeLists.txt (via FetchContent_MakeAvailable) as well as our port's CMakeLists.txt. Frankly i didn't even know they had a top level CMakeLists.txt - perhaps that is new.

For simplicity you can use our `FreeRTOS_Kernel_Import.cmake`, but basically you just want to add_subdirectory our CMakeLists.txt not theirs

Use link library (in your case) of `FreeRTOS-Kernel-Heap4`

Their build is build a static library without the right dependencies which is gonna be all kinds of wrong.

lurch, 2024-09-02:
You might also find https://github.com/raspberrypi/pico-examples?tab=readme-ov-file#freertos a helpful starting point?

sbasu3, 2024-09-08:
@kilograham ,  I have also been stuck on this . Getting error as : 
`error: 'sio_hw' undeclared (first use in this function); did you mean 'scb_hw'?`

Attaching my CMakeLists.txt
[CMakeLists.txt](https://github.com/user-attachments/files/16922820/CMakeLists.txt)

kilograham, 2024-09-08:
> @kilograham , I have also been stuck on this . Getting error as : `error: 'sio_hw' undeclared (first use in this function); did you mean 'scb_hw'?`
> 
> Attaching my CMakeLists.txt [CMakeLists.txt](https://github.com/user-attachments/files/16922820/CMakeLists.txt)

I do know what that one is; it is a bug if `pico_multicrore` is not included. not sure when it appeared.

If you could say exactly what you are stuck on; FreeRTOS branch/source, and SDK version that woud help

sbasu3, 2024-09-09:
Hi @kilograham , here are my details:

- FreeRTOS-Kernel : Main `a045081f73533ca972f0d9f8e23bf3ea61709b7d`
- pico-sdk : Main `efe2103f9b28458a1615ff096054479743ade236` , tag 2.0.0

Had to clean up my CMakeLists.txt nad now it is working. I am running on single core only .Here is the cleaned up version.
[CMakeLists.txt](https://github.com/user-attachments/files/16925499/CMakeLists.txt)

behines, 2024-09-10:
> You might also find https://github.com/raspberrypi/pico-examples?tab=readme-ov-file#freertos a helpful starting point?

Hi all, thanks for the tips.  I have been working with my kludged-together build for a few days to get a delivery out the door, but am back to seeing if I can de-kludge my setup.

I have previously found and tried the FreeRTOS example and I feel like I'm missing something.  CMake configure gives a bunch of non-fatal warnings ("No project() command is present", "cmake_minimum_required() should be called", etc.), then:

```
[cmake] System is unknown to cmake, create:
[cmake] Platform/PICO to use this system, please post your config file on discourse.cmake.org so it can be added to cmake
[cmake] Your CMakeCache.txt file was copied to CopyOfCMakeCache.txt. Please post that file on discourse.cmake.org.
[cmake] CMake Error at CMakeLists.txt:22 (pico_add_extra_outputs):
[cmake]   Unknown CMake command "pico_add_extra_outputs".

```

Of course I can make attempts to fix these things, but I feel like I'm missing something fundamental - I assume the example is supposed to Just Work.

behines, 2024-09-10:
> So it seems the majority of the issue is with the way your are importing FreeRTOS-Kernel. You are using their top level CMakeLists.txt (via FetchContent_MakeAvailable) as well as our port's [...]

Thanks again for the advice.  I am trying to implement these suggestions but it's not going well.  Is there a "just use this" CMakeLists.txt file somewhere?  The below is what I have right now, but configure fails with:

```
[cmake] CMake Error at FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/FreeRTOS_Kernel_import.cmake:57 (message):
[cmake]   Directory
[cmake]   'C:/Users/brad/Documents/PlanetA/Software/blink/build/FreeRTOS-Kernel' does
[cmake]   not contain an RP2040 port here: portable/ThirdParty/GCC/RP2040
```

I am considering an alternate approach which is to use basically the CMakeLists.txt that I had originally, but using FetchContent_Populate instead of FetchContent_MakeAvailable.

Below is the CMakeLists.txt where I tried to implement your suggestions.  Have I included them properly?  I have FreeRTOS in the FreeRTOS-kernel folder of my project.

Thanks,
Brad

```
cmake_minimum_required(VERSION 3.13)
set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

####
# VS Code Pico Extension created this section
#

# Initialise pico_sdk from installed location
# (note this can come from environment, CMake cache etc)
# == DO NEVER EDIT THE NEXT LINES for Raspberry Pi Pico VS Code Extension to work ==
if(WIN32)
    set(USERHOME $ENV{USERPROFILE})
else()
    set(USERHOME $ENV{HOME})
endif()
set(sdkVersion 2.0.0)
set(toolchainVersion 13_2_Rel1)
set(picotoolVersion 2.0.0)
include(${USERHOME}/.pico-sdk/cmake/pico-vscode.cmake)
# ====================================================================================

####
# Board and project
#

set(PICO_BOARD adafruit_itsybitsy_rp2040 CACHE STRING "Board type")
include(pico_sdk_import.cmake)
project(blink C CXX ASM)
pico_sdk_init()

####
# FreeRTOS config
#

add_library(freertos_config INTERFACE)
target_include_directories(freertos_config SYSTEM
INTERFACE
    include
)

target_compile_definitions(freertos_config
  INTERFACE
    projCOVERAGE_TEST=0
    ${definitions}
)
target_compile_options(freertos_config INTERFACE ${options})

####
# FreeRTOS Pico port
#

set( FREERTOS_HEAP "4" CACHE STRING "" FORCE)
set( FREERTOS_PORT "GCC_RP2040" CACHE STRING "" FORCE)
set (FREERTOS_KERNEL_PATH "FreeRTOS-Kernel")
set (RP2040_PORT_PATH ${FREERTOS_KERNEL_PATH}/portable/ThirdParty/GCC/RP2040 CACHE STRING "FreeRTOS RP2040 Port Path" FORCE)
#add_subdirectory(${FREERTOS_KERNEL_PATH})
add_subdirectory(${RP2040_PORT_PATH})
include(${RP2040_PORT_PATH}/FreeRTOS_Kernel_import.cmake)

####
# My app
#

add_executable(blink
    blink.c
    )

target_link_libraries(blink 
  FreeRTOS-Kernel-Heap4
  pico_stdlib
)

if (PICO_CYW43_SUPPORTED)
    target_link_libraries(blink pico_cyw43_arch_none)
endif()

pico_add_extra_outputs(blink)

```

kilograham, 2024-09-10:
```
####
# FreeRTOS config
#

add_library(freertos_config INTERFACE)
target_include_directories(freertos_config SYSTEM
INTERFACE
    include
)

target_compile_definitions(freertos_config
  INTERFACE
    projCOVERAGE_TEST=0
    ${definitions}
)
target_compile_options(freertos_config INTERFACE ${options})

####
# FreeRTOS Pico port
#

set( FREERTOS_HEAP "4" CACHE STRING "" FORCE)
set( FREERTOS_PORT "GCC_RP2040" CACHE STRING "" FORCE)
set (FREERTOS_KERNEL_PATH "FreeRTOS-Kernel")
set (RP2040_PORT_PATH ${FREERTOS_KERNEL_PATH}/portable/ThirdParty/GCC/RP2040 CACHE STRING "FreeRTOS RP2040 Port Path" FORCE)
#add_subdirectory(${FREERTOS_KERNEL_PATH})
add_subdirectory(${RP2040_PORT_PATH})
include(${RP2040_PORT_PATH}/FreeRTOS_Kernel_import.cmake)
```

Get rid of all of that

Make sure FREERTOS_KERNEL_PATH is correctly defined, and do

```
include(FreeRTOS_Kernel_import.cmake)
```

kilograham, 2024-09-10:
Any by correctly, i don't mean "set (FREERTOS_KERNEL_PATH "FreeRTOS-Kernel")" which as per your output is looking as a subdirectory of the build directory

perhaps "set (FREERTOS_KERNEL_PATH "${CMAKE_CURRENT_LIST_DIR}/FreeRTOS-Kernel")

behines, 2024-09-10:
> Any by correctly, i don't mean "set (FREERTOS_KERNEL_PATH "FreeRTOS-Kernel")" which as per your output is looking as a subdirectory of the build directory
> 
> perhaps "set (FREERTOS_KERNEL_PATH "${CMAKE_CURRENT_LIST_DIR}/FreeRTOS-Kernel")

Thanks, I'll give this a try.  Again, I really appreciate the super-fast responses, amazingly helpful.

While I was away for a few minutes, I tried the FetchContent_Populate approach.  This seems to work.  I was able to eliminate all of the "kludge section".  I bet I'm close to what you are suggesting.  I'll tidy up what I've got and post it tomorrow.

romain145, 2025-01-16:
I'm following up on this.
Building with the VS Code Pico extension 0.17.2, the SDK 2.1.0 and this [FreeRTOS Kernel](https://github.com/raspberrypi/FreeRTOS-Kernel/commit/4f7299d6ea746b27a9dd19e87af568e34bd65b15)

This is what I added to the CMakeLists.txt:
``` cmake
# Add FreeRTOS
set(FREERTOS_KERNEL_PATH ${CMAKE_CURRENT_LIST_DIR}/FreeRTOS-KernelV11.1.0)
include(FreeRTOS_Kernel_import.cmake)
include_directories(. ${FREERTOS_KERNEL_PATH}/include ${FREERTOS_KERNEL_PATH}/portable/ThirdParty/GCC/RP2040/include)
```

The include_directories is needed to fix 3 errors:
`fatal error: FreeRTOS.h: No such file or directory`
`fatal error: FreeRTOSConfig.h: No such file or directory`
`FreeRTOS-KernelV11.1.0/include/portable.h:53:14: fatal error: portmacro.h: No such file or directory`

This is the result now:
```
In file included from FreeRTOS-KernelV11.1.0/include/FreeRTOS.h:58,
                 from noplop.c:1:
FreeRTOS-KernelV11.1.0/portable/ThirdParty/GCC/RP2040/include/portmacro.h: In function 'vPortRecursiveLock':
FreeRTOSConfig.h:412:9: warning: implicit declaration of function 'taskDISABLE_INTERRUPTS'; did you mean 'portDISABLE_INTERRUPTS'? [-Wimplicit-function-declaration]
  412 |         taskDISABLE_INTERRUPTS(); \
      |         ^~~~~~~~~~~~~~~~~~~~~~
FreeRTOS-KernelV11.1.0/portable/ThirdParty/GCC/RP2040/include/portmacro.h:221:5: note: in expansion of macro 'configASSERT'
  221 |     configASSERT( ulLockNum < portRTOS_SPINLOCK_COUNT );
      |     ^~~~~~~~~~~~
[16/94] Building C object CMakeFiles/main.dir/main.c.obj
In file included from FreeRTOS-KernelV11.1.0/include/FreeRTOS.h:58,
                 from main.c:26:
FreeRTOS-KernelV11.1.0/portable/ThirdParty/GCC/RP2040/include/portmacro.h: In function 'vPortRecursiveLock':
FreeRTOSConfig.h:412:9: warning: implicit declaration of function 'taskDISABLE_INTERRUPTS'; did you mean 'portDISABLE_INTERRUPTS'? [-Wimplicit-function-declaration]
  412 |         taskDISABLE_INTERRUPTS(); \
      |         ^~~~~~~~~~~~~~~~~~~~~~
FreeRTOS-KernelV11.1.0/portable/ThirdParty/GCC/RP2040/include/portmacro.h:221:5: note: in expansion of macro 'configASSERT'
  221 |     configASSERT( ulLockNum < portRTOS_SPINLOCK_COUNT );
      |     ^~~~~~~~~~~~
```

Also, can you explain this directive that is used on some examples? It did not seem to make a difference here. Thanks
`set(FREERTOS_PORT "GCC_RP2040" CACHE STRING "" FORCE)`

peterharperuk, 2025-01-16:
@romain145 you are commenting on a closed issue. It looks a bit like your build is messed up. Can you try building the example in pico-examples? https://github.com/raspberrypi/pico-examples/tree/master/freertos

kilograham, 2025-01-16:
you should not have to add any include directories manually;

make your however that you are adding `target_link_libraries` for your target to `FreeRTOS-Kernel` or one of the derived libraries (see below for where the libraries are defined) - that is how the headers will show up

[code]
add_library(FreeRTOS-Kernel-Static INTERFACE)
target_compile_definitions(FreeRTOS-Kernel-Static INTERFACE
        configSUPPORT_STATIC_ALLOCATION=1
        )

target_sources(FreeRTOS-Kernel-Static INTERFACE ${CMAKE_CURRENT_LIST_DIR}/idle_task_static_memory.c)
target_link_libraries(FreeRTOS-Kernel-Static INTERFACE FreeRTOS-Kernel)

add_library(FreeRTOS-Kernel-Heap1 INTERFACE)
target_sources(FreeRTOS-Kernel-Heap1 INTERFACE ${FREERTOS_KERNEL_PATH}/portable/MemMang/heap_1.c)
target_link_libraries(FreeRTOS-Kernel-Heap1 INTERFACE FreeRTOS-Kernel)

add_library(FreeRTOS-Kernel-Heap2 INTERFACE)
target_sources(FreeRTOS-Kernel-Heap2 INTERFACE ${FREERTOS_KERNEL_PATH}/portable/MemMang/heap_2.c)
target_link_libraries(FreeRTOS-Kernel-Heap2 INTERFACE FreeRTOS-Kernel)

add_library(FreeRTOS-Kernel-Heap3 INTERFACE)
target_sources(FreeRTOS-Kernel-Heap3 INTERFACE ${FREERTOS_KERNEL_PATH}/portable/MemMang/heap_3.c)
target_link_libraries(FreeRTOS-Kernel-Heap3 INTERFACE FreeRTOS-Kernel)

add_library(FreeRTOS-Kernel-Heap4 INTERFACE)
target_sources(FreeRTOS-Kernel-Heap4 INTERFACE ${FREERTOS_KERNEL_PATH}/portable/MemMang/heap_4.c)
target_link_libraries(FreeRTOS-Kernel-Heap4 INTERFACE FreeRTOS-Kernel)

add_library(FreeRTOS-Kernel-Heap5 INTERFACE)
target_sources(FreeRTOS-Kernel-Heap5 INTERFACE ${FREERTOS_KERNEL_PATH}/portable/MemMang/heap_5.c)
target_link_libraries(FreeRTOS-Kernel-Heap5 INTERFACE FreeRTOS-Kernel)
[/code]

lurch, 2025-01-17:
> Building with the VS Code Pico extension 0.17.2, the SDK 2.1.0 and this [FreeRTOS Kernel](https://github.com/raspberrypi/FreeRTOS-Kernel/commit/4f7299d6ea746b27a9dd19e87af568e34bd65b15)

I see that you've linked to the "downstream" https://github.com/raspberrypi/FreeRTOS-Kernel repo. AFAIK the intention is that you should now be using the "upstream" repo, see e.g. https://github.com/raspberrypi/pico-feedback/issues/442

carlosdelfino, 2025-02-07:
After many attempts, starting with reading the README.md of the portable GCC_RP2040, and then using various methods on the internet, and without any success in compiling, I ended up deciding to come here to ask for help, and I found this issue that reports the same problems I had. I have success in compiling, for example, the portable GCC_ARM_CM0, but it crashes when creating the tasks. As for the RP2040, I have tried other approaches suggested on the web, but without success, and the last one that I came closest to something concrete is below.

Please, could someone guide me on how to correctly compile the portable for GCC_RP2040, I need to finish my TCC this week and I don't know what else to do.

```
# Generated Cmake Pico project file

cmake_minimum_required(VERSION 3.13)

set(PROJECT_NAME magnectostriccao)

set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

set(FREERTOS_KERNEL_PATH "${CMAKE_SOURCE_DIR}/FreeRTOS-Kernel" CACHE STRING "Path to the FreeRTOS Kernel")
set(FREERTOS_CONFIG_FILE_DIRECTORY "${CMAKE_SOURCE_DIR}/tasks/" CACHE STRING "") 

# Select the heap port.  values between 1-4 will pick a heap.
set(FREERTOS_HEAP "4" CACHE STRING "" FORCE)

#Select the native compile PORT
#set(FREERTOS_PORT "GCC_ARM_CM0" CACHE STRING "" FORCE)
#set(FREERTOS_PORT "GCC_RP2040" CACHE STRING "" FORCE)
#set(FREERTOS_PORT "ThirdParty_GCC_RP2040" CACHE STRING "" FORCE)

# Initialise pico_sdk from installed location
# (note this can come from environment, CMake cache etc)

# == DO NOT EDIT THE FOLLOWING LINES for the Raspberry Pi Pico VS Code Extension to work ==
if(WIN32)
    set(USERHOME $ENV{USERPROFILE})
else()
    set(USERHOME $ENV{HOME})
endif()
set(sdkVersion 2.1.0)
set(toolchainVersion 13_3_Rel1)
set(picotoolVersion 2.1.0)
set(picoVscode ${USERHOME}/.pico-sdk/cmake/pico-vscode.cmake)
if (EXISTS ${picoVscode})
    include(${picoVscode})
endif()
# ====================================================================================
set(PICO_BOARD pico_w CACHE STRING "Board type")

# Pull in Raspberry Pi Pico SDK (must be before project)
include(pico_sdk_import.cmake)

project(magnectostriccao C CXX ASM)

# Initialise the Raspberry Pi Pico SDK
pico_sdk_init()

# Add executable. Default name is the project name, version 0.1

add_executable(${PROJECT_NAME} main.c  
        inc/neopixel.c
        inc/ssd1306_i2c.c
        tasks/freeRTOS_helpers.c
        tasks/task_drone_control.c
        tasks/task_adc_with_dma.c
        tasks/task_display_oled.c
        tasks/task_tinyML.c 
)

pico_set_program_name(${PROJECT_NAME} "magnectostriccao")
pico_set_program_version(${PROJECT_NAME} "0.1")

# Generate PIO header
pico_generate_pio_header(${PROJECT_NAME} ${CMAKE_CURRENT_LIST_DIR}/inc/ws2818b.pio)
[... 179 lines cut ...]
In file included from /home/carlosdelfino/workspace/EmbarcaTech/magnectostric_freertos/FreeRTOS-Kernel/include/FreeRTOS.h:58,
                 from /home/carlosdelfino/workspace/EmbarcaTech/magnectostric_freertos/FreeRTOS-Kernel/stream_buffer.c:38:
/home/carlosdelfino/workspace/EmbarcaTech/magnectostric_freertos/FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/include/portmacro.h: In function 'vPortRecursiveLock':
/home/carlosdelfino/workspace/EmbarcaTech/magnectostric_freertos/tasks/FreeRTOSConfig.h:425:9: warning: implicit declaration of function 'taskDISABLE_INTERRUPTS'; did you mean 'portDISABLE_INTERRUPTS'? [-Wimplicit-function-declaration]
  425 |         taskDISABLE_INTERRUPTS(); \
      |         ^~~~~~~~~~~~~~~~~~~~~~
/home/carlosdelfino/workspace/EmbarcaTech/magnectostric_freertos/FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/include/portmacro.h:222:5: note: in expansion of macro 'configASSERT'
  222 |     configASSERT( ulLockNum < portRTOS_SPINLOCK_COUNT );
      |     ^~~~~~~~~~~~
[17/272] Building C object CMakeFiles/FreeRTOS-Kernel.dir/FreeRTOS-Kernel/tasks.c.o
In file included from /home/carlosdelfino/workspace/EmbarcaTech/magnectostric_freertos/FreeRTOS-Kernel/include/FreeRTOS.h:58,
                 from /home/carlosdelfino/workspace/EmbarcaTech/magnectostric_freertos/FreeRTOS-Kernel/tasks.c:39:
/home/carlosdelfino/workspace/EmbarcaTech/magnectostric_freertos/FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/include/portmacro.h: In function 'vPortRecursiveLock':
/home/carlosdelfino/workspace/EmbarcaTech/magnectostric_freertos/tasks/FreeRTOSConfig.h:425:9: warning: implicit declaration of function 'taskDISABLE_INTERRUPTS'; did you mean 'portDISABLE_INTERRUPTS'? [-Wimplicit-function-declaration]
  425 |         taskDISABLE_INTERRUPTS(); \
      |         ^~~~~~~~~~~~~~~~~~~~~~
/home/carlosdelfino/workspace/EmbarcaTech/magnectostric_freertos/FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/include/portmacro.h:222:5: note: in expansion of macro 'configASSERT'
  222 |     configASSERT( ulLockNum < portRTOS_SPINLOCK_COUNT );
      |     ^~~~~~~~~~~~
ninja: build stopped: subcommand failed.

 *  The terminal process "/home/carlosdelfino/.pico-sdk/ninja/v1.12.1/ninja '-C', '/home/carlosdelfino/workspace/EmbarcaTech/magnectostric_freertos/build'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

```

behines, 2025-02-09:
HI all,

I've been meaning for a long time to get back to this and write up what
worked for me, but this project fell off my plate.

Here's the CMakeLists.txt that finally worked for me, for my project
"Gen3".  Of course tune it to your own needs.  Hopefully this is helpful.

cmake_minimum_required(VERSION 3.13)

set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

### CMakeLists example: https://github.com/jondurrant/RPIPicoFreeRTOScpp

# Initialise pico_sdk from installed location
# (note this can come from environment, CMake cache etc)

# == DO NEVER EDIT THE NEXT LINES for Raspberry Pi Pico VS Code Extension
to work ==
if(WIN32)
    set(USERHOME $ENV{USERPROFILE})
else()
    set(USERHOME $ENV{HOME})
endif()
set(sdkVersion 2.0.0)
set(toolchainVersion 13_2_Rel1)
set(picotoolVersion 2.0.0)
include(${USERHOME}/.pico-sdk/cmake/pico-vscode.cmake)
#
====================================================================================

###################
# Pico setup
#

set(PICO_BOARD adafruit_itsybitsy_rp2040 CACHE STRING "Board type")

# Pull in Raspberry Pi Pico SDK (must be before project)
include(pico_sdk_import.cmake)

project(Gen3
  LANGUAGES C CXX ASM
  DESCRIPTION "Planet A Gen 3 collector"
)

# Initialise the Raspberry Pi Pico SDK
pico_sdk_init()

####################
# FreeRTOS
#

####
# Configurables

# Select Heap 4 and select the proper compiler for RP2040 (it's a Cortex
M0+)
#set(FREERTOS_HEAP "4" CACHE STRING "FreeRTOS Heap Type" FORCE)
#set(FREERTOS_PORT "GCC_RP2040" CACHE STRING "FreeRTOS port folder" FORCE)

####
# FreeRTOSConfig

# We put our FreeRTOSConfig.h in here
set(FREERTOS_CFG_DIRECTORY "include")

add_library(freertos_config INTERFACE)
target_compile_definitions(freertos_config
  INTERFACE
    projCOVERAGE_TEST=0
    ${definitions}
)
target_compile_options(freertos_config INTERFACE ${options})

# Specify the freertos_config "target".  It doesn't get built, is just
pulled in by other targets,
# so it is declared as an interface.
# "SYSTEM" says these are system headers, so ignore warnings.
target_include_directories(freertos_config SYSTEM
  INTERFACE
    ${FREERTOS_CFG_DIRECTORY}
)

####
# Fetch and build FreeRTOS

# Pull in FreeRTOS, fetching it from Github as needed.
include(FetchContent)  # Include FetchContent module

FetchContent_Declare(
  freertos_kernel
  GIT_REPOSITORY https://github.com/FreeRTOS/FreeRTOS-Kernel.git
  GIT_TAG        V11.1.0  # Use a specific tag or commit hash
)
FetchContent_Populate(freertos_kernel)  # Fetch and make FreeRTOS available

####
# The RP2040 port

# Print to verify the path (optional)
# Set the FREERTOS_KERNEL_PATH to the fetched content directory
# The RP2040 CMakeLists.txt uses this as well as us using it
set (FREERTOS_KERNEL_PATH ${freertos_kernel_SOURCE_DIR} CACHE STRING "FreeRTOS
Kernel Path" FORCE)
message(STATUS "FREERTOS_KERNEL_PATH is set to: ${FREERTOS_KERNEL_PATH}")
set (RP2040_PORT_PATH ${FREERTOS_KERNEL_PATH}/portable/ThirdParty/GCC/RP2040
CACHE STRING "FreeRTOS RP2040 Port Path" FORCE)
add_subdirectory(${RP2040_PORT_PATH})

# Pull in the folder with FreeRTOS_Kernel_import.cmake from the RP2040 port
by referencing it like this
# This adds the folder to the CMake search path for CMakeLists.txt files.
# This will cause it to process the CMakeLists.txt in the RP2040 folder
#add_subdirectory(${RP2040_PORT_PATH} FreeRTOS-RP2040)
include(${RP2040_PORT_PATH}/FreeRTOS_Kernel_import.cmake)

set(PICO_DEOPTIMIZED_DEBUG 1)

####################
# Our program
#

# Add executable. Default name is the project name, version 0.1
add_executable(Gen3
  TrackerMain.cpp
)

target_compile_definitions(Gen3 PUBLIC
  PICO_DEBUG_MALLOC=1
  PICO_CXX_DISABLE_ALLOCATION_OVERRIDES=1        # Tell the SDK that we
don't want its new and delete operators
  PICO_STDIO_USB_CONNECT_WAIT_TIMEOUT_MS=4000    # Wait up to 4 seconds for
USB stdio init to complete before proceeding.
  #PICO_USE_MALLOC_MUTEX=1
  #PICO_CXX_DISABLE_ALLOCATION_OVERRIDES=1
)

target_include_directories(Gen3 PUBLIC include)

pico_set_program_name(Gen3 "Gen3")
pico_set_program_version(Gen3 "0.1")

# Modify the below lines to enable/disable output over UART/USB
pico_enable_stdio_uart(Gen3 0)
pico_enable_stdio_usb(Gen3 1)
#pico_enable_stdio_semihosting(Gen3 1)

# Place the program in RAM (see
https://forums.raspberrypi.com/viewtopic.php?t=300974)
### When I did this, the USB serial port seemed to not work.
# pico_set_binary_type(Gen3 no_flash)

# Add the standard library to the build
#target_link_libraries(Gen3
#  pico_stdlib
#)

# Add the standard include files to the build
target_include_directories(Gen3 PRIVATE
  ${CMAKE_CURRENT_LIST_DIR}
  ${CMAKE_CURRENT_LIST_DIR}/.. # for our common lwipopts or any other
standard includes, if required
  #${PICO_SDK_PATH}/src/common/pico_time/include
  ${FREERTOS_KERNEL_PATH}/include
  ${RP2040_PORT_PATH}
  ${RP2040_PORT_PATH}/include
)

# Our PIO file that does the step pulse generation
pico_generate_pio_header(Gen3 ${CMAKE_CURRENT_LIST_DIR}/stepDirStepper.pio)

# Pico libraries that we require.  These take the place of Gen2's TI
DriverLib libraries
target_link_libraries(Gen3
  FreeRTOS-Kernel-Heap4
  hardware_i2c
  hardware_timer
  hardware_watchdog
  hardware_clocks
  hardware_adc
  hardware_dma
  hardware_pio
  stdc++             # C++ standard libraries
  pico_time
  pico_multicore
  pico_stdlib
)

# Produce artifacts needed by the debugger
pico_add_extra_outputs(Gen3)

kilograham, 2025-02-09:
I thought i had commented on this issue, but here is what a simple `CMakeLists.txt` should look life, assuming `sample.c`, `FreeRTOSConfig.h` and `FreeRTOS_Kernel_import.cmake` in the same directory as `CMakeLists.txt`

```
# The following CMake or environment variables should be defined:
# PICO_SDK_PATH = path/to/sdk
# FREERTOS_KERNEL_PATH = path/to/FreeRTOS-Kernel
cmake_minimum_required(VERSION 3.14)

# Pull in SDK (must be before project)
include(pico_sdk_import.cmake)

project(freertos-sample C CXX ASM)

set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)

# Initialize the SDK
pico_sdk_init()

# FREERTOS: include FreeRTOS Kernel libraries
include(FreeRTOS_Kernel_import.cmake)

add_executable(sample sample.c)

# FREERTOS: FreeRTOSConfig.h needs to be in the include path
target_include_directories(sample PRIVATE ${CMAKE_CURRENT_LIST_DIR})

# FREERTOS: Note, you should pick the FreeRTOS library that suits you best:
# 
#           FreeRTOS-Kernel-Heap1 thru FreeRTOS-Kernel_Heap4
#           or
#           FreeRTOS-Kernel-Static
target_link_libraries(sample
        pico_stdlib
        FreeRTOS-Kernel-Heap4
)

pico_add_extra_outputs(sample)
```

I will add a PR to update the README in FreeRTOS-Kernel and friends when I get the chance

rvodden, 2025-05-16:
For those fine individuals who stumble across this issue via google, I have a minimal CMakeLists.txt which works using FetchContent if you'd rather not use submodules:

```cmake
# Generated Cmake Pico project file

cmake_minimum_required(VERSION 3.13)

set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

# Initialise pico_sdk from installed location
# (note this can come from environment, CMake cache etc)

# == DO NOT EDIT THE FOLLOWING LINES for the Raspberry Pi Pico VS Code Extension to work ==
if(WIN32)
    set(USERHOME $ENV{USERPROFILE})
else()
    set(USERHOME $ENV{HOME})
endif()
set(sdkVersion 2.1.1)
set(toolchainVersion 14_2_Rel1)
set(picotoolVersion 2.1.1)
set(picoVscode ${USERHOME}/.pico-sdk/cmake/pico-vscode.cmake)
if (EXISTS ${picoVscode})
    include(${picoVscode})
endif()
# ====================================================================================

set(PICO_BOARD pico2_w CACHE STRING "Board type")
include(pico_sdk_import.cmake)

add_library(freertos_config INTERFACE)
target_include_directories(freertos_config INTERFACE ${CMAKE_SOURCE_DIR}/include)

include(FetchContent)
FetchContent_Populate(freertos_kernel
    GIT_REPOSITORY     "https://github.com/FreeRTOS/FreeRTOS-Kernel.git"
    GIT_TAG            "V11.2.0"
)
set(FREERTOS_KERNEL_PATH "${CMAKE_CURRENT_BINARY_DIR}/freertos_kernel-src" CACHE STRING "FreeRTOS Location" FORCE)

add_subdirectory(${FREERTOS_KERNEL_PATH}/portable/ThirdParty/Community-Supported-Ports/GCC/RP2350_ARM_NTZ)

project(blink C CXX ASM)

# Initialise the Raspberry Pi Pico SDK
pico_sdk_init()

add_executable(blink)
target_sources(blink PRIVATE src/blink.c)
target_compile_definitions(blink PUBLIC
    CYW43_LWIP=0
    WANT_HCI_DUMP=1
)
target_link_libraries(blink PUBLIC
    FreeRTOS-Kernel-Heap4
    freertos_config
    pico_stdlib
    pico_cyw43_arch_none
)

pico_enable_stdio_uart(blink 1)
pico_enable_stdio_usb(blink 0)

pico_add_extra_outputs(blink)
```