raspberrypi/pico-sdk #1662: Enable LWIP with no CYW43 device in SDK, e.g. using UART

rivimey, 2024-03-10:
I have wondered about using the rp2040 chip with IP networking, but using the UART ports (or usb-serial, the recent PIO-based Ethernet MII interface, or even USB-CDC/ENCM) rather than the CYW43 device. For me, a UART/PPP or UART/SLIP interface would be fine.

Currently the pico-sdk makes this relatively hard when using the pico-sdk because as far as I can see the code makes the assumption that the only possible way to use the lwip library is on a pico-w board and when using the cyw43 chip. Consequently all the IP-related code is littered with calls to cyw43 arch code and conditionals about there being a cyw43 device.

So, I'm asking if RaspberryPi could modify the pico-sdk so that use of cyw43 was just one possible method of IP networking. Such changes would I imagine be to do something like:

1. change build code that tests for CYW43 to test for a generic LWIP enabled flag.
2. set the LWIP enabled flag when using board: pico-w but ensure it can be enabled if wanted with board: pico too.
3. provide a way to specify the desired LWIP interface (e.g I'm using CYW43, or I'm using UART/PPP).
4. split out explicit calls to cyw43 arch code into either macros that can be redefined or an explicit glue layer.
5. Update cmake files appropriately.
6. ... probably other stuff! ...

To be clear I'm not asking for the other methods of using LWIP to be implemented, just asking that doing so doesn't require quite so much untangling of LWIP/CYW43-arch/CYW43-IP code. I don't know if it is practical or useful to think of multiple interfaces (e.g. a pico-w with a UART port, or an rp2040 using two UARTS)... it would be nice, but not necessary now.

I had wondered whether I might submit this as a PR but for a change of such wide scope I really want acceptance in principle from RPI to do this, as a minimum, and very preferably help doing it!

peterharperuk, 2024-03-11:
There are some changes in the pipeline. But I don't think there's anything stopping you using lwip in Pico builds. You obviously shouldn't be using any cyw43 libraries for Pico builds, but the lwip libraries are all separate as far as I'm aware.

kilograham, 2024-03-11:
see also https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_lwip/doc.h

lurch, 2024-03-11:
> see also https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_lwip/doc.h

... which is also readable at https://www.raspberrypi.com/documentation/pico-sdk/networking.html#pico_lwip if you're not a fan of reading raw Doxygen :wink: