raspberrypi/pico-sdk #1670: Add static version of queue

daveythacher, 2024-03-17:
I was looking into the queue API and I noticed that it works off the heap. It should be possible to implement this as static?
https://github.com/raspberrypi/pico-sdk/blob/6a7db34ff63345a7badec79ebea3aaef1712f374/src/common/pico_util/queue.c#L11-L18

It is a little more risky, but for people that do not want to use heap they are probably stuck with that.
```C
void static_queue_init_with_spinlock(queue_t *q, uint element_size, uint element_count, uint spinlock_num, uint8_t *storage) {
    lock_init(&q->core, spinlock_num);
    q->data = storage;
    q->element_count = (uint16_t)element_count;
    q->element_size = (uint16_t)element_size;
    q->wptr = 0;
    q->rptr = 0;
}
```

peterharperuk, 2024-03-18:
Yes, that would be good

lurch, 2024-03-18:
In your proposed new function, it might also be worth passing in a `storage_size` parameter, and then asserting that `storage_size >= ((element_count + 1) * element_size)` ?

daveythacher, 2024-03-18:
Forgot about the assert. I had considered it but did not see the point in changing the return signature.

Note there is no nullptr check.