raspberrypi/pico-sdk #1693: [question] Build fails for simple test program due to missing native compiler but pioasm or elf2uf2 are not used

HackXIt, 2024-05-02:
Greetings,

I have an [open issue on conan](https://github.com/conan-io/conan/issues/16193) which describes the following build error when using the pico SDK in a conan project:

<details>
  <summary>Build error</summary>

```
======== Calling build() ========
conanfile.py (pico-conan-blink/0.0.1): Calling build()
conanfile.py (pico-conan-blink/0.0.1): Running CMake.configure()
conanfile.py (pico-conan-blink/0.0.1): RUN: cmake -G "Unix Makefiles" -DCMAKE_TOOLCHAIN_FILE="/home/rini/git-stash/pico-conan-blink/build/Debug/generators/conan_toolchain.cmake" -DCMAKE_INSTALL_PREFIX="/home/rini/git-stash/pico-conan-blink" -DCMAKE_POLICY_DEFAULT_CMP0091="NEW" -DCMAKE_BUILD_TYPE="Debug" "/home/rini/git-stash/pico-conan-blink"
-- Using Conan toolchain: /home/rini/git-stash/pico-conan-blink/build/Debug/generators/conan_toolchain.cmake
Using PICO_SDK_PATH from environment ('/home/rini/pico/sdk')
PICO_SDK_PATH is /home/rini/pico/sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
-- Conan toolchain: C++ Standard 17 with extensions ON
-- Conan toolchain: Setting BUILD_SHARED_LIBS = OFF
-- The C compiler identification is GNU 12.2.1
-- The CXX compiler identification is GNU 12.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
Using PICO_SDK_PATH from environment ('/home/rini/pico/sdk')
PICO_SDK_PATH is /home/rini/pico/sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
PICO compiler is pico_arm_gcc
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc
Using PICO_SDK_PATH from environment ('/home/rini/pico/sdk')
PICO_SDK_PATH is /home/rini/pico/sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
PICO compiler is pico_arm_gcc
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc - broken
CMake Error at /usr/share/cmake-3.25/Modules/CMakeTestCCompiler.cmake:70 (message):
  The C compiler

    "/usr/bin/arm-none-eabi-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/rini/git-stash/pico-conan-blink/build/Debug/CMakeFiles/CMakeScratch/TryCompile-j07YgO
    
    Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_0f430/fast && /usr/bin/gmake  -f CMakeFiles/cmTC_0f430.dir/build.make CMakeFiles/cmTC_0f430.dir/build
    gmake[1]: Entering directory '/home/rini/git-stash/pico-conan-blink/build/Debug/CMakeFiles/CMakeScratch/TryCompile-j07YgO'
    Building C object CMakeFiles/cmTC_0f430.dir/testCCompiler.c.o
    /usr/bin/arm-none-eabi-gcc   -O3 -DNDEBUG -o CMakeFiles/cmTC_0f430.dir/testCCompiler.c.o -c /home/rini/git-stash/pico-conan-blink/build/Debug/CMakeFiles/CMakeScratch/TryCompile-j07YgO/testCCompiler.c
    Linking C executable cmTC_0f430
    /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_0f430.dir/link.txt --verbose=1
    /usr/bin/arm-none-eabi-gcc -O3 -DNDEBUG CMakeFiles/cmTC_0f430.dir/testCCompiler.c.o -o cmTC_0f430 
    /usr/lib/gcc/arm-none-eabi/12.2.1/../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/12.2.1/../../../arm-none-eabi/lib/libc.a(lib_a-exit.o): in function `exit':
    /build/newlib-afIbHz/newlib-3.3.0/build/arm-none-eabi/newlib/libc/stdlib/../../../../../newlib/libc/stdlib/exit.c:64: undefined reference to `_exit'
    collect2: error: ld returned 1 exit status
    gmake[1]: *** [CMakeFiles/cmTC_0f430.dir/build.make:99: cmTC_0f430] Error 1
    gmake[1]: Leaving directory '/home/rini/git-stash/pico-conan-blink/build/Debug/CMakeFiles/CMakeScratch/TryCompile-j07YgO'
    gmake: *** [Makefile:127: cmTC_0f430/fast] Error 2
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:30 (project)

-- Configuring incomplete, errors occurred!
[... 234 lines cut ...]
        tc.extra_cxxflags = ["--mcpu=cortex-m0plus", "-mthumb"]
        tc.preprocessor_definitions["PICO_SDK_PATH"] = "deps/pico-sdk"
        tc.generate()

    def build(self):
        get(self, "https://github.com/raspberrypi/pico-sdk/archive/refs/tags/1.5.1.tar.gz", destination="deps/pico-sdk", strip_root=True)
        common_cmake_args = []
        cmake = CMake(self)
        cmake.configure(cli_args=common_cmake_args)
        if self.options.parallel:
            common_cmake_args += ["--parallel"]
        cmake.build(cli_args=common_cmake_args)
```

</details>

Conan does not provide the native C/C++ compiler to the build process, but only `arm-none-eabi` cross-compiler.

Despite my CMakeLists.txt not using the cmake functions provided by the `pioasm` and `elf2uf2` tools, there still seems to be a hidden step that does require some sort of native compiler, resulting in the simple test program error.

Am I doing something wrong or forgetting something?

I've setup conan so that it loads the `pico_sdk_import.cmake` during the build step and it's visible in the logs that configuration catches multiple C/C++ compiler setups _(I am yet unsure why it is done multiple times, I suspect due to the additional import in my `CMakeLists.txt`)_

However, from there on, the project should actually compile just fine, since it has the cross-compiler, it finds the SDK, so I don't see a reason why it should fail in the way it does.

peterharperuk, 2024-05-02:
By coincidence we were talking about this today...

Yes, it's a common problem I'm afraid. Some people work around the issue https://forums.raspberrypi.com/viewtopic.php?t=350982

On windows - which doesn't have a native compiler - the following is included before including pico_sdk_import.cmake

```
set(Pioasm_TARGET Pioasm)
set(Pioasm_EXECUTABLE ${pico-sdk-tools_DIR}/pioasm.exe)
if(NOT TARGET ${Pioasm_TARGET})
    add_executable(${Pioasm_TARGET} IMPORTED)
endif()
set_property(TARGET ${Pioasm_TARGET} PROPERTY IMPORTED_LOCATION ${Pioasm_EXECUTABLE})
set(Pioasm_FOUND 1)

set(ELF2UF2_TARGET ELF2UF2)
set(ELF2UF2_EXECUTABLE ${pico-sdk-tools_DIR}/elf2uf2.exe)
if(NOT TARGET ${ELF2UF2_TARGET})
    add_executable(${ELF2UF2_TARGET} IMPORTED)
endif()
set_property(TARGET ${ELF2UF2_TARGET} PROPERTY IMPORTED_LOCATION ${ELF2UF2_EXECUTABLE})
set(ELF2UF2_FOUND 1)
```

HackXIt, 2024-05-03:
That's great news!

I have included your cmake snippet before the import, but it didn't seem to do the trick just yet.

I quickly applied my hack to get around the simple test program error which I can't seem to fix:
```
set(CMAKE_C_COMPILER_WORKS 1)
set(CMAKE_CXX_COMPILER_WORKS 1)
```

But now I get the following build error:
<details>
<summary>Build log</summary>

```
$ conan build . --profile=rp2040-debug

======== Input profiles ========
Profile host:
[settings]
arch=armv6
build_type=Debug
compiler=gcc
compiler.cppstd=gnu17
compiler.libcxx=libstdc++11
compiler.version=12
os=baremetal
[conf]
tools.cmake.cmaketoolchain:user_toolchain=['$ENV{PICO_SDK_PATH}/external/pico_sdk_import.cmake']
[buildenv]
CC=/usr/bin/arm-none-eabi-gcc
CXX=/usr/bin/arm-none-eabi-g++
AR=/usr/bin/arm-none-eabi-ar
AS=/usr/bin/arm-none-eabi-as
RANLIB=/usr/bin/arm-none-eabi-ranlib
LD=/usr/bin/arm-none-eabi-ld
STRIP=/usr/bin/arm-none-eabi-strip

Profile build:
[settings]
arch=armv8
build_type=Release
compiler=gcc
compiler.cppstd=gnu17
compiler.libcxx=libstdc++11
compiler.version=12
os=Linux

======== Computing dependency graph ========
Graph root
    conanfile.py (pico-conan-blink/0.0.1): /home/rini/git-stash/pico-conan-blink/conanfile.py

======== Computing necessary packages ========

======== Installing packages ========

======== Installing packages ========

======== Finalizing install (deploy, generators) ========
conanfile.py (pico-conan-blink/0.0.1): Calling generate()
conanfile.py (pico-conan-blink/0.0.1): Generators folder: /home/rini/git-stash/pico-conan-blink/build/Debug/generators
conanfile.py (pico-conan-blink/0.0.1): CMakeToolchain generated: conan_toolchain.cmake
conanfile.py (pico-conan-blink/0.0.1): Preset 'conan-debug' added to CMakePresets.json. Invoke it manually using 'cmake --preset conan-debug' if using CMake>=3.23
conanfile.py (pico-conan-blink/0.0.1): If your CMake version is not compatible with CMakePresets (<3.23) call cmake like: 'cmake <path> -G "Unix Makefiles" -DCMAKE_TOOLCHAIN_FILE=/home/rini/git-stash/pico-conan-blink/build/Debug/generators/conan_toolchain.cmake -DCMAKE_POLICY_DEFAULT_CMP0091=NEW -DCMAKE_BUILD_TYPE=Debug'
conanfile.py (pico-conan-blink/0.0.1): CMakeToolchain generated: CMakePresets.json
conanfile.py (pico-conan-blink/0.0.1): CMakeToolchain generated: ../../../CMakeUserPresets.json
conanfile.py (pico-conan-blink/0.0.1): Generating aggregated env files
conanfile.py (pico-conan-blink/0.0.1): Generated aggregated env files: ['conanbuild.sh', 'conanrun.sh']

======== Calling build() ========
[... 71 lines cut ...]
/tmp/cciwm6MN.s:86: Error: selected processor does not support `dmb' in ARM mode
/tmp/cciwm6MN.s:143: Error: selected processor does not support `dmb' in ARM mode
/tmp/cciwm6MN.s:168: Error: selected processor does not support requested special purpose register -- `msr PRIMASK,r0'
gmake[2]: *** [CMakeFiles/pico-conan-blink.dir/build.make:132: CMakeFiles/pico-conan-blink.dir/home/rini/pico/sdk/src/rp2_common/hardware_claim/claim.c.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
/tmp/ccoBBkCO.s: Assembler messages:
/tmp/ccoBBkCO.s:79: Error: selected processor does not support `dmb' in ARM mode
/tmp/ccoBBkCO.s:119: Error: selected processor does not support `dmb' in ARM mode
/tmp/ccoBBkCO.s:158: Error: selected processor does not support `dmb' in ARM mode
/tmp/ccoBBkCO.s:197: Error: selected processor does not support `dmb' in ARM mode
/tmp/ccoBBkCO.s:236: Error: selected processor does not support `dmb' in ARM mode
/tmp/ccoBBkCO.s:275: Error: selected processor does not support `dmb' in ARM mode
/tmp/ccoBBkCO.s:314: Error: selected processor does not support `dmb' in ARM mode
/tmp/ccoBBkCO.s:353: Error: selected processor does not support `dmb' in ARM mode
/tmp/ccoBBkCO.s:443: Error: selected processor does not support `dmb' in ARM mode
/tmp/ccoBBkCO.s:716: Error: selected processor does not support `dmb' in ARM mode
gmake[2]: *** [CMakeFiles/pico-conan-blink.dir/build.make:146: CMakeFiles/pico-conan-blink.dir/home/rini/pico/sdk/src/rp2_common/hardware_sync/sync.c.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1524: CMakeFiles/pico-conan-blink.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2

ERROR: conanfile.py (pico-conan-blink/0.0.1): Error in build() method, line 68
        cmake.build(cli_args=common_cmake_args)
        ConanException: Error 2 while executing
```
</details>

HackXIt, 2024-05-03:
I fixed the additional build errors by enforcing the target `-mcpu` and `-march` in the conan profile.

This did not occur to me beforehand, because these flags are **normally** set when using `pico_sdk_import.cmake` and `pico_sdk_init()`.

I will continue this in the conan issue, since the package manager introduces this weird behaviour.

Thanks for all the insight, there were definitly one or two steps necessary to get the core issue, would have taken me way longer without your snippet to avoid `pioasm` and `elf2uf2`.

closed 2024-05-03 by HackXIt