raspberrypi/pico-sdk #1697: USB stdio issue at lower sys clock

szlldm, 2024-05-05:
USB stdio fails if  SYS_CLK_KHZ overridden to 32MHz (PLL_SYS_VCO_FREQ_KHZ and PLL_SYS_POSTDIV1, 2 are also set to valid values).
/dev/ttyACM0 does not even appear
dmesg shows errors:
 usb 1-7.2: new full-speed USB device number 71 using xhci_hcd
 usb 1-7.2: Device not responding to setup address.
 usb 1-7.2: device descriptor read/all, error -110
Meanwhile user code on the Pico runs ok.
At 64MHz everything seems fine (only SYS_CLK_KHZ and PLL_SYS_POSTDIV2 changed).
Is this expected, or is anything i can do to make it work?
Thx

matsobdev, 2024-05-06:
They say in the datasheet, it NOPs when running higher clock than USB 48 MHz to wait for it, so maybe at lower clock it can not keep up. Maybe right, maybe wrong.

liamfraser, 2024-07-22:
The hardware has been designed to work at a clk_sys of >= clk_usb. Perhaps we could assert about this when initializing USB? Would need to think about where to put it as you can have clk_sys as slow as you like if you are not using USB. Any thoughts @kilograham. Also worth noting this in the datasheet

liamfraser, 2024-07-22:
I have added a note about this to the datasheet which will come out in the next release

liamfraser, 2024-08-01:
We will add an assertion to check the clk_sys frequency is at least 48MHz when using USB. However, it's quite hard to defend against this post clocks initialization as we'd have to put an assert in every clock_configure function to check it thoroughly.