raspberrypi/pico-sdk #1698: pico_flash should depend on hardware_exception

jaguilar, 2024-05-07:
Hi there, I think pico_flash should depend on hardware_exception? It unconditionally includes hardware_exception, and if I try to compile a file that depends on pico_flash without editing the SDK, I get this compile error:

```
[build] /home/jaguilar/projects/pico-sdk/src/rp2_common/pico_flash/flash.c:8:10: fatal error: hardware/exception.h: No such file or directory
[build]     8 | #include "hardware/exception.h"
[build]       |          ^~~~~~~~~~~~~~~~~~~~~~
```

If I add hardware_exception to the dependency list of pico_flash then everything works as expected.

kilograham, 2024-05-18:
it doesn't seem like the file actually needs the header on a quick check

jaguilar, 2024-05-19:
@kilograham Adjusted the pull request to remove the unneeded includes. Added a compilation test to show the before and after state. I don't know if you prefer to have tests like that in the tree. Let me know if not and I can remove it.

kilograham, 2024-06-05:
merged into develop

PR #1699 (2024-06-05): pico_flash stops including certain unneeded headers
Fixes #1698
closed 2024-06-05 by kilograham